Opentrons/opentrons #5810: bug: hitting start run button too soon causes app to lose communication on subsequent pause

nickcrider, 2020-06-03:
# Overview

Hitting the "Start Run" button before the robot has finished homing after the final labware calibration step, causes the App to lose connection with the robot, and requires power-cycling the robot to resolve.

# Steps to reproduce

1. upload any valid protocol with a pause step to the robot
2. proceed through labware calibration as normal
3. On the last step when the UI says "Return tip and proceed to run" get your fast clicking finger ready
4. Click "Return Tip..." and then quickly click "Start Run" As soon as the button pops up in the app
5. Try to cancel before the pause or resume after the pause
6. you now need to powercycle the robot in order to reconnect via the app

# Current behavior
No abnormal errors can be seen in the `journalctl` logs

[image]= 
[image]
Closing and reopening the app looks like a websocket error
[image]

# Expected behavior

The speed at which start run is clicked after completing labware calibration does not effect subsequent operation of the robot.

nickcrider, 2020-06-03:
https://github.com/Opentrons/opentrons/issues/5239 is likely related

mcous, 2020-06-04:
IIRC, the return tip action (from a robot perspective) tends to remain in progress even as the UI switches over to the run page. The screenshots provided seem to imply that the API is shutting down the websocket (or it is otherwise crashing) if the run RPC command arrives while the previous return tip RPC command is still processing

amitlissack, 2020-06-07:
There appears to be insufficient protection from creating multiple sessions at the same time.  I think we need a mutex rather than just a state variable (ie `self._session_lock`).

amitlissack, 2020-06-08:
I think this is the root cause...

Session.run does not inhibit running while a run is active. Session.run will call Session._run which tries to acquire the motion lock. If a protocol run is active, then the run app will be stuck waiting for the second Session.run to return, because we are blocked on acquiring of motion lock. 

If the protocol has a pause step, the robot is totally deadlocked because both RPC threads are blocked. 

Solution, I think, is to not allow Session.run while the session is in Session._run.

@mcous  @SyntaxColoring  what do you think?

sfoster1, 2020-06-08:
With something like using an actual grown-up mutex for the session lock and calling `try_acquire`?

@mcous would we have to swallow the error on the robot side or can the app handle an error response in that case

mcous, 2020-06-08:
@amitlissack oh boy, that sounds fun

@sfoster1 the app is resilient to the `session.run` call raising an error. It will (IIRC) cause an error banner to be displayed in the app. The client still needs to do a better job of disabling buttons when an action should not be allowed, but that's orthogonal to this fix

nickcrider, 2020-06-09:
Just verified that this also occurs in protocols without a pause step, based on a few new customer reports.

nickcrider, 2020-06-09:
I just verified based on some additional customer reports that in protocols without a pause step similar symptoms are seen (unresponsive to pause commands from app, locked up in the "spinner" on cancel) but the app is able to recover communication once the protocol has finished running on the robot.

commit d8c63d7b (2020-06-16): fix(robot-server, api): reject commands while robot is moving (#5878)

* Modification to ThreadedAsyncLock. It now runs in two modes: regular lock and forbid. Lock is normal mutex locking. Forbid will raise an error if trying to acquired the already acquired lock. This allows us to reject commands while a long running command is running eliminating deadlocks.
Also, use a Lock instead of Rlock. Rlock was not achieving locking in our async code as all calls are coming an the same thread.
* use forbid lock for /robot/home and /robot/move.
* redo use of ThreadedAsyncLock in api.
rename the decorat [... cut]
closed 2020-06-16 by amitlissack