Opentrons/opentrons #5829: bug: Simulation fails when protocol contains a tuberack with height greater than 118 mm

ethanfjones, 2020-06-05:
# Overview

Loading certain standard Opentrons labware and using GEN2 single or multi pipettes creates a `LabwareHeightError` even in simulation with the default deck calibration and robot settings.

# Current behavior
A `LabwareHeightError` is thrown for the following combinations of pipettes and labware:

**Pipette**
`p300_single_gen2`
`p100_single_gen2`
`p20 _single_gen2`
`p300_multi_gen2`
`p20_multi_gen2`

**Labware**
`opentrons_10_tuberack_falcon_4x50ml_6x15ml_conical`
`opentrons_10_tuberack_nest_4x50ml_6x15ml_conical`
`opentrons_15_tuberack_falcon_15ml_conical`
`opentrons_15_tuberack_nest_15ml_conical`
`opentrons_6_tuberack_falcon_50ml_conical`
`opentrons_6_tuberack_nest_50ml_conical`

Other combinations are certainly possible were not exhaustively tried at this time. We might need to develop a test to watch for this. 

# Steps to reproduce

With opentrons package 3.18.1, simulate the following protocol via opentrons_simulate:

```python
from opentrons import protocol_api

metadata = {
    'apiLevel': '2.4'
}

def run(protocol:protocol_api.ProtocolContext):

    # labware
    tuberack500 = protocol.load_labware(
        'opentrons_10_tuberack_falcon_4x50ml_6x15ml_conical', '1') 
    wellplate48 = protocol.load_labware(
        'corning_48_wellplate_1.6ml_flat', '2') 
    tiprack = protocol.load_labware(
        'opentrons_96_tiprack_1000ul', '4') 

    
    left_pipette = protocol.load_instrument(
        'p1000_single_gen2', 'left', tip_racks=[tiprack]) 

 
    left_pipette.transfer(400, tuberack500.wells_by_name()['A1'],wellplate48.wells_by_name()['A1']) 
```
# Expected behavior

You should be able to simulate any standard Opentons labware with any pipette.

nickcrider, 2020-06-05:
Related https://github.com/Opentrons/opentrons/issues/5774

ethanfjones, 2020-06-12:
Workaround is to use known good deck calibration and robot settings files and store them locally on your /.opentrons directory.

# Steps to resolve 
1. Download [calibration.zip](https://github.com/Opentrons/opentrons/files/4770799/calibration.zip)

2. Extract files 
3. Place them in your /.opentrons directory

PR #5908 (2020-06-18): fix(api): Simulate protocols with a realistic deck transform
## overview

Closes #5774 and closes #5829. Support noticed that the labware height error would often get thrown during simulation for certain combinations of pipettes + labware. This is because simulation does not actually use a valid deck transform for any operation checks.

If the labware height error is going to be useful outside of the robot, we should simulate using a valid transform.

Making this a draft PR because I wanted to get feedback.

## changelog

- Modify tests to get them passing
- Add in a valid deck transform to si [... cut]
closed 2020-06-18 by Laura-Danielle