Opentrons/opentrons #5846: bug: newline char is mishandled in robot logs

nickcrider, 2020-06-08:
# Overview

The update to 3.18.0 (and likely the introduction of the new http backend) caused the logs downloaded through the app to have the newline character rendered as the string literal `\n`

# Current behavior

This has not been fixed as of 3.19.0-alpha.0

# Steps to reproduce

`curl -X GET "http://{ROBOT_IP}:31950/logs/api.log?format=text&records=50000" -H "accept: application/json"` or download the logs through the app

you'll see a bunch of `\n`'s where there should be newlines

# Expected behavior

Logs are readable without `sed`-ing the newline char.

sfoster1, 2020-06-09:
Well that's awful

SyntaxColoring, 2020-06-09:
To clarify what's going on here, I think the logs are being encoded in one big JSON string. So it's "correct" that they have a bunch of internal newlines, in an unhelpful sense.

The pedantically correct way to decode it is something like:

```shell
python -c "import json; import sys; print(json.load(open(sys.argv[1])))" ~/Downloads/api.log
```

Which accounts for other escape sequences like `\t` and `\\`.

PR #5850 (2020-06-09): fix(robot-server): preserve newlines in text logs
The fastapi automatic encoder for responses was doing a bit too much
when we returned a raw string in the logs response. By using the
Response class directly, we can specify the encoding as text/plain if
we're returning text logs instead of letting fastapi infer it.

Closes #5846

## Testing
Install it and get logs - there should be newlines
commit 6420b5fa (2020-06-09): fix(robot-server): preserve newlines in text logs (#5850)

The fastapi automatic encoder for responses was doing a bit too much
when we returned a raw string in the logs response. By using the
Response class directly, we can specify the encoding as text/plain if
we're returning text logs instead of letting fastapi infer it.

Closes #5846
closed 2020-06-09 by sfoster1