Opentrons/opentrons #5903: feat: Provide comprehensive release notes for intermediate versions on the way to the latest version

SyntaxColoring, 2020-06-12:
# Overview

If your Opentrons App isn't already near the latest version, and you try to update, our hand-written [release notes][release-notes] will omit a lot of changes that might be relevant to you.

## Details of current behavior

At the time of writing, the latest release is v3.18.1. If you try to update from v3.14.1, the Opentrons App will only tell you about the changes from v3.17.1 to v3.18.1.

I believe the app just shows the latest [release-notes.md][release-notes]. This contains a nice hand-written summary of changes, but *only between the latest minor versions of the latest two releases.* For example, the changes between v3.17.1 to v3.18.1. It will *not* comprehensively represent the changes between v3.17.0 to v3.18.1, nor from v3.14.1 to v3.18.1.

The [CHANGELOG.md][changelog] file contains a full history, but it's generally too low-level to be useful to customers.

# Acceptance criteria

* [ ] When you update the Opentrons App from any version to any newer version, it shows you *all* the hand-written release notes that are relevant.
* [ ] Bonus points: You can also view the release notes online.
  * A file on GitHub would be sufficient.
  * We can kind of already do this by viewing [release-notes.md][release-notes] on GitHub. The problem is that we can't see the whole history as a single stream. We have to flip between release tags.

It would be fine if both of these improvements were only supported for versions going forward, not also retroactively.

[changelog]: app-shell/CHANGELOG.md
[release-notes]: app-shell/build/release-notes.md

mcous, 2020-06-13:
On the app side, we use [remark](https://github.com/remarkjs/remark) to process the markdown in release-notes.md for rendering, which internally is a very nicely structured pipeline where the Markdown is parsed into a syntax tree before being rendered to HTML.

A simple solution (or starting point for a solution) could be to try making `release-notes.md` prepend only, with a horizonatal rule node (`---`) between each "entry". The UI side can then split on those nodes, render the first entry by default, and hide the reset behind a "click to expand"

This should work for both the App release notes and the Robot OS release notes

mcous, 2021-06-03:
We've started leaving the entries for old versions in the release notes

closed 2021-06-03 by mcous