Opentrons/opentrons #5927: How to connect to the OT-2 via serial connections to send my own G-code

zachaayuna, 2020-06-16:
Hello, I am using the opentron to write my own G-code driver and was wondering what the best way would be to connect to the OT-2

nickcrider, 2020-06-16:
Hi there @zachaayuna thanks for writing!

The OT-2 uses a [smoothie motor controller](http://smoothieware.org/), and that's connected to the robot's RaspberryPI via a USB-serial connection.

Here's our implementation of the G-code driver https://github.com/Opentrons/opentrons/blob/edge/api/src/opentrons/drivers/smoothie_drivers/driver_3_0.py

But, would you mind sharing a bit more about what you're trying to do and why you want to write your own G-code driver for the robot? I unfortunately don't think I have enough info from you to be of much help right now.

zachaayuna, 2020-06-16:
So we are not using it to pipette we want to change the head and control the machine using our own G-code driver or maybe make adjustments to yours to suite our needs. I was just trying to directly connect to the smoothieboard so I can customize the movements of the machine.

zachaayuna, 2020-06-16:
I don't know if what I am saying makes sense but basically I am trying to directly connect to the smoothieboard via python

zachaayuna, 2020-06-16:
Or more simply, How would I be able to run my own driver on the OT-2

zachaayuna, 2020-06-16:
Would I need to take the outer shell off and plug directly into the board? We want to be able to control the arm in very specific ways.

theosanderson, 2020-06-16:
(Am fellow user not dev).
In theory you can send GCODE with `driver._send_comand` https://github.com/Opentrons/opentrons/blob/edge/api/src/opentrons/drivers/smoothie_drivers/driver_3_0.py#L980 It's worth thinking about what you want to command it to do that you can't do by commanding the gantry to move to a particular position, e.g.

```protocol._hw_manager.hardware.move_to(
            types.Mount.RIGHT, position, speed=speed)```

nickcrider, 2020-06-16:
Thanks @theosanderson ! You took the words right from my keyboard!

@zachaayuna, my advice is that you spend some time considering the recommendation from our user above. The API does an awful lot of work to help transform the smoothie coordinates into "deck coordinates" that correspond to physical positions on the robot deck. I don't want to doubt your abilities, but I also want to make sure you don't waste a bunch of time reinventing the wheel.

But despite that, if you still decide that you'd like to do it your own way (it's opensource after all, more power to you!), you can [start taking apart the robot](https://support.opentrons.com/en/articles/3460245-accessing-the-motor-controller-board), [access the relevant pins on the smoothie](http://smoothieware.org/uart), and drive it however you want.

Hope that helps!

zachaayuna, 2020-06-17:
Ok thank you @nickcrider If I wanted to use some of your functionality

zachaayuna, 2020-06-17:
Can I use the opentron api to move to exact positions by sending coordinates

theosanderson, 2020-06-17:
Yes, (untested:)
```
from opentrons import types
protocol._hw_manager.hardware.move_to(
            types.Mount.RIGHT, types.Point(100,100,100))
```

zachaayuna, 2020-06-18:
Works perfectly Thank you! @theosanderson

zachaayuna, 2020-06-22:
Hi @theosanderson ,

Is there a way to move the one Z axis without the other coming back up to starting position

Thank you

theosanderson, 2020-06-22:
You could write your own version of `move_to` without `cache_and_maybe_retract_mount` https://github.com/theosanderson/opentrons/blob/edge/api/src/opentrons/hardware_control/__init__.py#L711 or move down to a lower level command looking at `move_to` for inspiration.

zachaayuna, 2020-06-22:
Ok thanks

zachaayuna, 2020-06-23:
@theosanderson whe i ssh to the robot do i change the python files in data/packages/usr/local/lib/python3.6/site-packages/opentrons or where can I enter the functions on robots rasPi

zachaayuna, 2020-06-23:
I have just been playing around with the machine to understand functionality and and trying some things out and just had some questions and you 
@theosanderson seem very knowledgeable on the topic so thank you

theosanderson, 2020-06-23:
You don't want to change the opentrons library itself. Write your own functions that call on some of its functions. Here is some code where I've done that. https://github.com/theosanderson/tube_checkout/blob/master/tube_checkout/tube_checkout/tube_movement.py You can do that in Jupyter for example, especially for prototyping

theosanderson, 2020-06-23:
So write your own `move_to` but keep it in your own code

zachaayuna, 2020-06-23:
I understand and then you just run that as protocol then

theosanderson, 2020-06-23:
Yes or you can run it via SSH / Jupyter

zachaayuna, 2020-06-23:
Okay thanks

> 
> 
> Yes or you can run it via SSH / Jupyter

-- 

*NOTICE:*
The information contained in this e-mail message is intended 
for the exclusive
use of the addressee(s) and may contain proprietary 
information. Any
proprietary information contained herein should be handled 
per executed NDAs.
If you are not the intended recipient of this message, 
you are hereby notified
that any form of dissemination of this 
communication is strictly prohibited .
Please note that any views or 
opinions presented in this email are solely those
of the author and do not 
necessarily represent those of the company. Finally,
the recipient should 
check this email and any attachments for the presence of
viruses. The 
company accepts no liability for any damage caused by any virus
transmitted 
by this email. AAYUNA Inc, Suite# 306, 7540 Windsor Drive,
Allentown, PA, 
USA, www.aayuna.com <http://www.aayuna.com>

closed 2020-07-28 by nickcrider