Opentrons/opentrons #5970: bug (app): Unable to find robots on 3.19.0-alpha.2

dozercodes, 2020-06-22:
# Overview

The app fails to locate robots to connect. Instead the app is stuck perpetually looking for robots

# Current behavior

[image]

App fails to find robot to connect to on either Wifi or USB
# Steps to reproduce

1) Install build 3.19.0-alpha.2 (issue confirmed on windows and mac builds)
2) Open the Opentrons app to the Robot tab
3) Turn on your robot
4) Observe

# Expected behavior

App should find your robot to connect to.

dozercodes, 2020-06-22:
tagging @sfoster1 for visibility

mcous, 2020-06-23:
Closed by #5971. For clarity, this was not an issue with robot discovery, but rather, and issue with the app's renderer thread crashing due to an uncaught exception

PR #5971 (2020-06-22): refactor(app): fix race condition introduced by async anayltics init
## overview

#5874 switched the app's MixPanel module from sync loading at launch to async loading when the app-shell sent over the config. This introduced a race condition where the app's previously existing logic to change the user's opt-in status in response to user initiated events would erroneously fire at config init, because it saw a change from `null` to "opted in or out".

This PR ensures the runtime opt-in change only fires after the config is fully loaded to avoid opt-in changes conflicting with (or rather, happeni [... cut]
closed 2020-06-23 by mcous