Opentrons/opentrons #5986: refactor(api,robot-server): move `opentrons.calibration` from api to robot-server

amitlissack, 2020-06-23:
# Overview

The calibration flows belong in the `robot-server`. Reasons:
- The actions in the flow are exclusively driven by HTTP requests. 
- The flow code would benefit from using models and constants defined in robot-server and vice versa
- The calibration flow is not meant to be used by customers except via HTTP requests. Hence no need for it to be in distributed package.
- Isolate pydantic to robot-server

# Acceptance criteria

- [ ] `opentrons.calibration` is moved to `robot-server`

SyntaxColoring, 2020-06-25:
Would this affect factory (CLI) calibration?

We currently use this command to launch the tool:

```
python -m opentrons.deck_calibration.dc_main
```

amitlissack, 2020-06-26:
> Would this affect factory (CLI) calibration?
> 
> We currently use this command to launch the tool:
> 
> ```
> python -m opentrons.deck_calibration.dc_main
> ```

It will not. It will only affect new calibration flows (ie cal check, tip length cal, etc).

PR #6012 (2020-06-29): refactor(robot-server, api): Robot server move calibration
# Overview

Move calibration check and tip length calibration flows to robot-server fro api

# Changelog

Created `robot_server.robot` package and moved `opentrons.calibration` into it.
Integrated `robot_server.robot.calibration` into robot-server
removed pydantic as dependency in API

# Review requests

Do you approve of `robot_server.robot` as a top level package?

# Risk assessment

Nearly None. Just a moving of code.

closes #5986
commit 07af9818 (2020-06-29): refactor(robot-server, api): Robot server move calibration (#6012)

* move calibration from api to robot-server

* integrate robot_server.robot.calibration into project.

* remove pydantic from api dependencies

closes #5986
closed 2020-06-29 by amitlissack