Opentrons/opentrons #5991: bug: pipette won't move to the next available tip

ffffffrank, 2020-06-23:
# Overview

I'm using python script to control other Instruments along with OT2 (as remote server), I have the python script in OT2 to call from the local computer via SSH.

stdin, stdout, stderr = ssh_client.exec_command('sh -lc "python3 /test.py pipette.pick_up()"')
stdin, stdout, stderr = ssh_client.exec_command('sh -lc "python3 /test.py pipette.drop()"')
stdin, stdout, stderr = ssh_client.exec_command('sh -lc "python3 /test.py pipette.pick_up()"')
stdin, stdout, stderr = ssh_client.exec_command('sh -lc "python3 /test.py pipette.drop()"')

For the first picking up, it started picking at A1 in the tip rack. But for the 2nd pick up, it still went to A1, instead of going to the next available tip.

Any suggestion would be high appreciated.

# Steps to reproduce

# Current behavior

# Expected behavior

nickcrider, 2020-06-23:
Hi @ffffffrank 

I think this is because of the way you are structuring the calls through SSH. Each time you do this, the API creates a new protocol context object and thus resets the tip tracking logic. To take advantage of the tip tracking logic you'll have to use the same protocol `ProtocolContext` for all of your motion calls.

To quote the API documentation
> `ProtocolContext` is designed to be ephemeral. The lifetime of a particular instance should be about the same as the lifetime of a protocol.

You'll find some related info here as well: https://docs.opentrons.com/v2/new_advanced_running.html

Hope this helps!

ffffffrank, 2020-06-23:
> Hi @ffffffrank
> 
> I think this is because of the way you are structuring the calls through SSH. Each time you do this, the API creates a new protocol context object and thus resets the tip tracking logic. To take advantage of the tip tracking logic you'll have to use the same `ProtocolContext` for all of your motion calls.
> 
> To quote the API documentation
> 
> > `ProtocolContext` is designed to be ephemeral. The lifetime of a particular instance should be about the same as the lifetime of a protocol.
> 
> You'll find some related info here as well: https://docs.opentrons.com/v2/new_advanced_running.html
> 
> Hope this helps!

Hi Nick,

Thanks for the quick reply. Do you mean that I need to process all the command in one line shell?

nickcrider, 2020-06-23:
Yep! I think that would fix the issue you are seeing with the tip tracking.

closed 2020-07-23 by sfoster1