Opentrons/opentrons #6010: bug: labware.wells() returns a single member list

nickcrider, 2020-06-25:
# Overview

`labware.wells(0)` returns a strictly typed, single member list that causes a `TypeError` when passed to a function expecting a `Well` or `Point` 

https://github.com/Opentrons/opentrons/blob/6e247262e65da6eb8a5567944e52839ebbc4e7d9/api/src/opentrons/protocol_api/labware.py#L571

This confuses users, and is not clearly explained in the docs.

# Current behavior

`pipette.aspirate(vol, plate.wells(0)` returns `TypeError`
`pipette.aspirate(vol, plate.wells()[0])` works.

This confuses users who are novice programmers and are struggling with our API for the first time, as it relies on intuitive knowledge of a programming concept. They look almost equivalent, and should, in a sane, ideal and _pythonic_ world, work the same.

It's [buried in the docs that you probably shouldn't do this](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.labware.Labware.wells) (likely for this very reason) but folks unfortunately don't read that far before contacting support.

# Expected behavior

when `labware.wells()` is supplied with one argument, it should return the `Well` rather than a `list`

Laura-Danielle, 2020-06-25:
This is behavior we were trying to move away from for API v2, and maybe it's time we finally remove it completely. From the docstring:

```
Note that this method takes args for backward-compatibility, but use
of args is deprecated and will be removed in future versions. Args
can be either strings or integers, but must all be the same type (e.g.:
`self.wells(1, 4, 8)` or `self.wells('A1', 'B2')`, but
`self.wells('A1', 4)` is invalid.
```

@sfoster1 @umbhau do we feel like we're ready to take these things out of API v2?

nickcrider, 2020-06-25:
I am fine with either, I think the real problem is when trying what seems reasonable `pipette.aspirate(vol, plate.wells(0))` you get an error like this
```
location should be a Well or Location, but it is [A5 of NEST 12 Well Reservoir 15 mL on 3]
```

And if it's your first time figuring out what those `[]` thingies mean, you're gonna get really frustrated

amitlissack, 2020-11-17:
I vote for removing the args to `Labware.wells`.

SyntaxColoring, 2024-02-20:
Superseded by RSS-475.

closed 2024-02-20 by SyntaxColoring