Opentrons/opentrons #6015: bug: importing robot tricks the protocol parsing logic into thinking a malformed v2 protocol is v1

nickcrider, 2020-06-26:
# Overview

Importing robot tricks the protocol parsing logic into thinking a malformed v2 protocol is v1. I am not sure if this is technically a bug but it did confuse me and a user.

# Steps to reproduce
Write a malformed v2 protocol that imports robot but forgets `metadata`
```python
from opentrons import robot
def run(protocol):
    pipette1 = protocol.load_instrument('p1000_single_gen2', 'left', tip_racks=nonexistant)
    pipette1.pick_up_tip()
```
This simulates successfully, so when uploaded to the robot the robot will say it has "no steps". 

However, this:
```python
def run(protocol):
    pipette1 = protocol.load_instrument('p1000_single_gen2', 'left', tip_racks=nonexistant)
    pipette1.pick_up_tip()
```

Raises the expected `RuntimeError`
```
RuntimeError: If this is not an API v1 protocol, you must specify the target api level in the apiLevel key of the metadata. For instance, metadata={"apiLevel": "2.0"}
```
# Expected behavior

A malformed protocol like the above should raise a `RuntimeError` about the `apiLevel`

nickcrider, 2020-06-26:
Perhaps this will be fixed in 3.19.0 by https://github.com/Opentrons/opentrons/pull/5859 ? I've only tested on 3.18.1

nickcrider, 2020-07-07:
This seems to have gotten slightly more confusing after 3.19.0 since all of the prints were removed from `opentrons_simulate`

sfoster1, 2020-07-23:
The thing is, that first protocol is a valid (if not particularly useful) APIv1 protocol. APIv1 protocols can be almost anything, and could absolutely include `run` functions.

sfoster1, 2020-07-23:
We can't change the detection machinery for this I think, so I'm closing in favor of #6015

closed 2020-07-23 by sfoster1