Opentrons/opentrons #6023: feat: PD Pipette/tip change- Air gap

Kadee80, 2020-06-29:
# Implementation Details
- Air gap volume will scale with pipette changes the same way Disposal volume does

shlokamin, 2020-08-11:
Is this not covered by #6264 and #6007 (done via clamping the air gap volume when the pipette/tiprack changes)? Or is this something else?

IanLondon, 2020-08-12:
Marking WIP bc requirements need fleshing out -- does `Volume Per Well` affect the air gap volume in the same way that it affects disposal volume (disposal volume is only for multi-dispense. Max disposal volume = pipette_and_tip_capacity - 2 * volume, because higher than that would break up the multi-dispense into single transfers)

If so, that would apply to multi-aspirate and multi-dispense, but not for single transfers? (bc single transfers can be any volume, if it's too big it'll get broken up into multiple single transfer)

Confusing to compare to disposal volume bc
- Air gap's max is different (max air gap = pipette_and_tip_capacity - pipette_minimum_aspirate_volume)
- Disposal volume is only for multi-dispense, no other paths

closed 2020-08-26 by shlokamin