Opentrons/opentrons #6039: Connecting Via serial

zachaayuna, 2020-07-01:
I have accessed the smoothieboard and have found the TX RX pins. If I connect to those pins can I communicate to the robot via serial as that is what TX RX pins usually do. I can't find the answer to this in the documentation so I thought I would ask here

zachaayuna, 2020-07-01:
And would you happen to know if it takes 5V or 3.3 V for the connection

nickcrider, 2020-07-01:
Hi @zachaayuna !

We recommend you use the FTDI debug header on the Pi GPIO shield, rather than the TX-RX directly on the smoothie. You should see a little switch that you can flip to 'debug' and then connect a standard 3.3V 6pin FTDI UART cable. That way you don't have to mess with the ribbon cable connecting the smoothie to the power distribution board.

When using the debug header, you'll want to wait until the Pi has fully booted before switching over to the debug header, as the startup process of the API server tries to talk to the smoothie, and if it cant connect, the boot process on the Pi hangs.

Hope this helps!

closed 2020-07-23 by sfoster1