Opentrons/opentrons #6045: bug: robot.set_lights() not working in v3.19.0

gfleetwood, 2020-07-02:
# Overview

# Steps to reproduce

I upgraded the robot's firmware and hardware from the app, and the opentrons package via pip: `pip install opentrons --upgrade`. Once I did that this code no longer switches the lights on and off:

```
from opentrons import robot

robot.connect()
robot.set_lights(rails = 1)
robot.set_lights(rails = 0)
```

gfleetwood, 2020-07-08:
This is time sensitive, so is there any way to do a rollback to the previous version of the firmware and software?

theosanderson, 2020-07-09:
(fellow user not a dev). It could be worth trying running this on the command line before your protocol: `systemctl stop opentrons-robot-server`. If it works then you can run it in your protocol with subprocess.

gfleetwood, 2020-07-09:
@theosanderson Thanks for the tip. I'm actually just using a python script that is intermittently triggered every couple of hours. I think I need the server because the process involves making a put request to an endpoint but I'm not sure. Should I do something like this?

```
from opentrons import robot
import os

robot.connect()

os.system("systemctl stop opentrons-robot-server")
robot.set_lights(rails = 1)
os.system("systemctl start opentrons-robot-server")

# code for running experiment and making http request

os.system("systemctl stop opentrons-robot-server")
robot.set_lights(rails = 0)
os.system("systemctl start opentrons-robot-server")
```

theosanderson, 2020-07-09:
Essentially, but I think you need to stop the server before you import robot. In fact 
```
import os
os.system("systemctl stop opentrons-robot-server")

robot.set_lights(rails = 0)
from opentrons import robot

robot.connect()

os.system("systemctl start opentrons-robot-server")

```

Might just work with the rest of the code as normal (but it also might not). If your PUT request is to get a webcam image there are also alternative ways to do this at the command line FWIW.

gfleetwood, 2020-07-09:
@theosanderson I'm using the command line through os.system to take images with the internal camera which I then send to the endpoint. Combining this with the other parts of the pipeline leans more towards python scripting. Hopefully your advice will sidestep the light issue. Thanks!

theosanderson, 2020-07-09:
OK -- if that endpoint isn't on the robot server itself then it shouldn't matter. Good luck regardless!

gfleetwood, 2020-07-11:
@theosanderson So the code didn't work for me. I know you're not an employee but do you know of anything else to try? 

Also, are the lights different somehow on the latest version? Looking at them more closely yesterday there were 3 (4 if there's a set at the back) sets of small, circular lights. These didn't respond to the code, but I also suspect these aren't the main ones since they don't turn off at all (even after unplugging the robot) and are really dim. Previously it was obvious when the light was one so wherever the main light is it wasn't responding to the code.

theosanderson, 2020-07-11:
So I don't know exactly why you're having this issue, the simplest way to guarantee to fix it is to upgrade to start using the V2 API: https://docs.opentrons.com/v2/new_advanced_running.html . The reason you've hit this is that the V1 API is not the main focus of attention nowadays so bugs are more likely in it. Would re-writing the code to use that take too long? If you do go for that then you need to stop the robot server before the first time you connect to the robot.

theosanderson, 2020-07-11:
The lights aren't different - they are big LED strips near the top of the robot

gfleetwood, 2020-07-11:
Right. LED Strips. Completely fluffed the term. Even though they were "on" yesterday it was much dimmer than usual. Does robot.set_lights just set them to be much brighter? 

To be honest, I thought of switching to the V2 API in anticipation of problems with an old version, but couldn't get any of the examples to run. The scripts on the page you posted look complete but take this one from their examples page:

```
from opentrons import protocol_api

metadata = {'apiLevel': '2.2'}

def run(protocol: protocol_api.ProtocolContext):
    plate = protocol.load_labware('corning_96_wellplate_360ul_flat', 1)
    tiprack_1 = protocol.load_labware('opentrons_96_tiprack_300ul', 2)
    p300 = protocol.load_instrument('p300_single', 'right', tip_racks=[tiprack_1])

    p300.transfer(100, plate['A1'], plate['B1'])
```

It gives no indication of how to actually run the code, so I tried (what I thought was) the obvious thing:

```
run(protocol_api.ProtocolContext)
```

And that threw an error complaining about "_api_level" or something. I don't remember the exact wording. To honest, I find the V2 API to be more opaque and the docs aren't that great (for both APIs). Continuing with the V1 API was the lesser of two evils.

OK, so I should try this or should the stop and start calls be in different places?

```
import os
import opentrons.execute
from opentrons import protocol_api

def run(protocol: protocol_api.ProtocolContext):
    # the contents of your protocol are here...

os.system("systemctl stop opentrons-robot-server")

protocol = opentrons.execute.get_protocol_api('2.0')

os.system("systemctl start opentrons-robot-server")

protocol.set_rail_lights(on = True)
run(protocol) 
protocol.set_rail_lights(on = False)
```

And this speaks to the opaqueness I mentioned before. I have the protocol object created from `opentrons.execute` but then its type signature in the run function comes from an import that I'm not directly using. Is it just being imported and added to the global namespace behind the scenes?

theosanderson, 2020-07-11:
Writing quickly on the go: the examples are only for protocols submitted
through the OT app. If you are running directly through Python follow the
example on that advanced control page and keep stuff in the main flow not
in a run function. I think if you try to extend the Advanced Control e.g.
it will start to make sense.

gfleetwood, 2020-07-11:
Great. Will see if that works. Thanks!

theosanderson, 2020-07-11:
It's safest to stop the server before you even import opentrons (indeed I
think that may be essential)

PR #6091 (2020-07-13): fix(api): fix APIv1 GPIO ctl for cmdline and jupyter protocol execution

# Overview
When the robot server is running, protocols executed in Jupyter or using `opentrons_execute` would not have control over the GPIOs. Currently, a warning with a workaround is shown if the protocol is in APIv2. However, there is no indication given to the user of why GPIO controls are not working when they are running a v1 protocol.

This PR fixes the issue by surfacing the same warning as v2 and actually implementing the workaround for v1 protocols.

closes #6045

# Changelog
- adds a new `info` message right be [... cut]
commit 09b64594 (2020-07-13): fix(api): fix APIv1 GPIO ctl for cmdline and jupyter protocol execution (#6091)

closes #6045
closed 2020-07-13 by ahiuchingau