Opentrons/opentrons #6095: bug: nonexistent docs pages redirect the browser infinitely instead of 404ing

SyntaxColoring, 2020-07-09:
# Overview

Trying to visit a page in the API documentation that doesn't exist throws the browser into a redirect loop.

# Steps to reproduce

Try to visit a URL that doesn't exist. Examples:

* https://docs.opentrons.com/v2/apiv2index.html (a wrong URL found on [a Protocol Library page](https://protocols.opentrons.com/protocol/illumina-nextera-XT-library-prep-part3))
* https://docs.opentrons.com/v2/this_page_does_not_exist
* https://docs.opentrons.com/v2/this_page_does_not_exist.html
* https://docs.opentrons.com/v1/this_page_also_does_not_exist
* https://docs.opentrons.com/v2/index (works if you add `.html` to the end)

# Current behavior

The page refreshes over and over again, with https://docs.opentrons.com/v2/v2 shown in the address bar.

Also, if you click one of the links to the v1 or v2 docs, it doesn't break you out of the loop. It just adds a `/v1` or `/v2` to the URL shown in the address bar.

# Expected behavior

Trying to visit a nonexistent page should show a 404 error.

# See also

* #4672, the fix for which we seem to have regressed on?

asogluizzo, 2020-12-04:
examine docs

mcous, 2022-04-11:
Closes by #9918