Opentrons/opentrons #6103: spike(robot-server): python protocols can import user defined python

amitlissack, 2020-07-10:
Protocols are so verbose. So much repetition 

How do we enable this?

Possible approaches: 
- use python that is included in uploaded bundle/zip. Execute it and include it globally
- allow saving py files to the robot file system. save files in a directory structure that makes it possible to import a protocol from another protocol.

theosanderson, 2020-07-10:
This would be really cool. If you go with a uploaded bundle, then it still feels like the nicest way to interact with this would still be with pythonic `import`s? That seems likely to be achievable and would keep it consistent with Jupyter.

mcous, 2020-07-21:
Proof of concept working, incorporating buildout into Run Sessions

closed 2020-07-21 by mcous

amitlissack, 2020-07-21:
Branch showing prototype: https://github.com/Opentrons/opentrons/tree/robot-server_protocol_session_proto