Opentrons/opentrons #6125: bug: When calibrating from the bottom, an image is sometimes shown of calibrating to the top of the well

Zanecrc1, 2020-07-14:
# Overview

# Steps to reproduce

# Current behavior

# Expected behavior

Zanecrc1, 2020-07-14:
In the application during the calibration step an image of the "calibrate to top" step will show when the setting to calibrate to the bottom is checked. The application will still say to calibrate to the bottom, only the wrong image is shown

SyntaxColoring, 2020-07-15:
Hm, yeah, that sounds wrong.

Could you post a protocol file that causes this behavior? It might have to do with the exact labware it loads.

Zanecrc1, 2020-07-15:
> ***CAUTION:*** This email is not from a BCM Source. Only click links or open attachments you know are safe.

Looks like this is a known issue

https://support.opentrons.com/en/articles/4195424-one-pipette-is-slightly-off-position-during-protocol-runs

The protocol is a bit sensitive so i've removed all of the steps minus just whats required to upload the script.

Zanecrc1, 2020-07-15:
All sensitive bits removed. 

`from opentrons import types

metadata={"apiLevel": "2.5"}

def run(protocol_context):

	# Pipette Tip Setup
	p300rack1 = protocol_context.load_labware('opentrons_96_tiprack_300ul', '6')
	p300rack2 = protocol_context.load_labware('opentrons_96_tiprack_300ul', '1')
	p300rack3 = protocol_context.load_labware('opentrons_96_tiprack_300ul', '4')
	p300rack4 = protocol_context.load_labware('opentrons_96_tiprack_300ul', '7')
	p300rack5 = protocol_context.load_labware('opentrons_96_tiprack_300ul', '11')
	#p300rack6 = protocol_context.load_labware('opentrons_96_tiprack_300ul', '8')

	p20rack1 = protocol_context.load_labware('opentrons_96_tiprack_20ul', '10')

	p300 = protocol_context.load_instrument('p300_multi', 'right')
	p20 = protocol_context.load_instrument('p20_multi_gen2', 'left')

	# Module setup
	magdeck = protocol_context.load_module('magnetic module gen2', '9')
	mag_plate = magdeck.load_labware('generic_96_wellplate_800ul')
	tempdeck = protocol_context.load_module('temperature module gen2', '3')
	#transfer_plate = protocol_context.load_labware('generic_96_wellplate_800ul', '3')
	elution_plate = tempdeck.load_labware('opentrons_96_aluminumblock_biorad_wellplate_200ul')

	#labware setup
	reagent_wells = protocol_context.load_labware('nest_12_reservoir_15mL', '2')

	#reagent location setup
	rna_buffer = reagent_wells['A1'] 
	wash1 = reagent_wells['A2'] #Zymo wash solution 2
	wash2 = reagent_wells['A3'] #Zymo wash solution 1
	eth = reagent_wells['A4'] #ethanol, 100%
	eluant = reagent_wells['A5']

	#volume setup
	magbead_mixvol = 150 #ul, volume to use when mixing reactions involving magnetic beads
	wash_mixvol = 75 #ul, volume to use when mixing during a wash step
	mixdepth = -13 #mm from top/bottom, the depth at which the pipette tip accesses the well on mixing
	dispense_depth = 10 #mm from bottom, the depth at which the pipette tip accesses the well on dispensing
	aspirate_depth = 1 #mm from bottom, the depth at which the pipette tip accesses the well on aspiration

	elution_vol = 8 #ul, the volume of eluant to use on the final elution post SPRI clean-up

	# Maximum height the magnets in the magnetic module can be raised to on the loaded labware 
	maxplateheight = 11

	#Robot movement tuning
	protocol_context.max_speeds['A'] = 50 # speed in mm/s of the pipettor head up and down 

	#Sample Setup- only use row A for multichannel, the whole column will be used. 
	samples_1 = 'A1', #'A3', 'A5', 'A7', 'A9', 'A11'
	samples_2 = 'A2', #'A4', 'A6', 'A8', 'A10', 'A12' 

	tips_all = 'A1', 'A2', 'A3', 'A4', 'A5', 'A6', 'A7', 'A8', 'A9', 'A10', 'A11', 'A12'
	tips_A = 'A1', 'A2', 'A3', 'A4', 'A5', 'A6'
	tips_B = 'A7', 'A8', 'A9', 'A10', 'A11', 'A12'

	#Pre-protocol actions/preparations
	magdeck.disengage()
`

SyntaxColoring, 2020-07-20:
See also #4734

closed 2025-12-15 by sfoster1