Opentrons/opentrons #6135: feat: make Sphinx warnings CI errors

SyntaxColoring, 2020-07-15:
# Overview

This ticket takes over for PR #5794, which was a first step at a solution to this problem.

`make docs` prints a lot of Sphinx warnings, which makes it hard to work on the API docs. (Legitimate mistakes are easy to lose in the noise.)

*After fixing those warnings,* new warnings should be turned into CI errors, so they're caught early and won't accumulate again.

# Implementation details

# Design

# Acceptance criteria

* Sphinx warnings cause CI builds to fail, preventing offending PRs from being merged.

SyntaxColoring, 2021-12-08:
#9016 re-enabled non-fatal warnings because of surprising/spurious/uncontrollable failures in CI. We think they were because CI Python env changes made some internal Sphinx thing or plugin raise warnings, and Sphinx couldn't differentiate them from legitimate problems in our doc sources. We should revisit after updating Sphinx—we're currently on v2.2.0 and the latest is v4.

ecormany, 2022-07-19:
we've updated to Sphinx 5.0.1 (#10583) but still getting tons of warning text in the logs, e.g. https://github.com/Opentrons/opentrons/runs/7325998045

SyntaxColoring, 2022-10-12:
Replaced with Jira ticket RTC-198.

closed 2022-10-12 by SyntaxColoring