Opentrons/opentrons #6140: feat(api): Include self-resolution guidance for multichannel transfer error

nickcrider, 2020-07-15:
# Overview

When writing PAPIv2 protocols with 384 well plates, users who haven't bumped their `apiLevel` to 2.2 or greater will hit an error when they attempt to access the second row of the plate (#4678).

https://github.com/Opentrons/opentrons/blob/8271b0fb96d9dc1ddddd09aec90ca42329958656/api/src/opentrons/protocol_api/transfers.py#L786

To resolve this, users writing new protocols just need to set the correct apiLevel. The error raised should say this.

# Design

A proposed error message follows
```
RuntimeError: Invalid {id} for multichannel transfer: {old_well_list}. 
If this is a valid 384-well plate, try using API Version 2.2 or greater.
```

sfoster1, 2020-07-17:
I think this may not be the root of the error - do we have a minimal reproducing case protocol?

The line called out here I think only executes in API version 2.2 or greater and is intended to catch cases where the protocol requests a transfer including row 3 or subsequent of the plate.

nickcrider, 2020-07-17:
Good catch! This ticket was opened in error.

Looking at my test case again, #5085 is the underlying issue.

closed 2020-07-17 by nickcrider