Opentrons/opentrons #6161: bug: custom 384 well plates can't be used with the complex commands

nickcrider, 2020-07-16:
# Overview

The complex API level commands e.g. `transfer()` and `distribute()` will allow multichannel access to the 2nd row of the standard Opentrons 384-well plate, but will not allow this for a custom defined 384-well plate.

# Steps to reproduce

Create a custom 384 well plate with the web tool https://labware.opentrons.com/create/ . Or use this one
[Greiner Bio-one 384 Well Plate 30 µL.zip](https://github.com/Opentrons/opentrons/files/4934700/Greiner.Bio-one.384.Well.Plate.30.L.zip)

Now try to use it with `transfer()` into the B row.

```python
def run(protocol):

    custom_348 = protocol.load_labware('greinerbioone_384_wellplate_30ul', 11)
    ot_384 = protocol.load_labware('corning_384_wellplate_112ul_flat', 10)

    tiprack_1 = protocol.load_labware('opentrons_96_tiprack_20ul', 1)
    p20_multi = protocol.load_instrument('p20_multi_gen2', 'left', tip_racks=[tiprack_1])
    
    # this works
    p20_multi.transfer(7.5, ot_384["B3"], ot_384["B14"])

    # this line will throw an error
    p20_multi.transfer(7.5, custom_348["B3"], custom_348["B14"])
```

# Current behavior

The labware creator tool produces labware definitions with `"format": "irregular"` for all labware. This appears to be the root cause. Changing to `"format": "384standard"` in the custom labware definition resolves the issue.

# Expected behavior

Complex commands should perform the same on custom labware as Opentrons standard labware.

SyntaxColoring, 2020-07-16:
I think this is a dupe of https://github.com/Opentrons/opentrons/issues/5085, right?

(I just skimmed these two issues. There might be a difference that I glossed over.)

nickcrider, 2020-07-16:
Ah yep it is. Good Call!

Thanks Github search ☹️

SyntaxColoring, 2020-07-16:
Oh oh, wait, let me [try something](https://docs.github.com/en/github/managing-your-work-on-github/about-duplicate-issues-and-pull-requests)...

SyntaxColoring, 2020-07-16:
Duplicate of #5085

SyntaxColoring, 2020-07-16:
Ooooooh!

SyntaxColoring, 2020-07-16:
I have no idea what marking an issue as a duplicate like that actually does.

Okay, I've had my fun.

closed 2020-07-16 by SyntaxColoring