Opentrons/opentrons #6171: bug: Labware Library definition for nest 2mL deep well plate

bndo, 2020-07-17:
# Overview
Our labware definition for the NEST 96 Deepwell Plate 2mL states that it has a `U-bottom` it should be updated to match [NEST's product pages](http://www.cell-nest.com/page94?product_id=101&_l=en). 

# Steps to reproduce

https://labware.opentrons.com/nest_96_wellplate_2ml_deep/

# Current behavior

Our definition currently states `U-bottom` instead of `V-bottom`.

# Expected behavior

Both the text and the image need to be updated to be a `V-bottom`

IanLondon, 2020-07-20:
Blocked until we have labware versioning :(

IanLondon, 2020-07-27:
After convo, we decided that it's OK to mutate the definition because we're only changing metadata (`groups[i].metadata.wellBottomShape`)

commit 134029dd (2020-07-27): fix(shared-data): fix well bottom shape of nest_96_wellplate_2ml_deep

Closes #6171
PR #6216 (2020-07-28): fix(shared-data): fix well bottom shape of nest_96_wellplate_2ml_deep
# Overview

Closes #6171

# Changelog

- changes `nest_96_wellplate_2ml_deep` wellBottomShape from U to V, matching the real labware

# Review requests

- Compare LL to production, should show a V-bottom not U-bottom
- This `wellBottomShape` field truly is only metadata: it's only used for display (specifically, only in LL) and does not affect behavior. Right?

# Risk assessment

Low. Mutates a labware definition, but only mutates a metadata field.
commit 63505112 (2020-07-28): fix(shared-data): fix well bottom shape of nest_96_wellplate_2ml_deep (#6216)

Closes #6171
closed 2020-07-28 by IanLondon