Opentrons/opentrons #6172: bug: Consecutive 'Pause' commands skip to last command for protocol modal

tanaes, 2020-07-17:
# Overview

I'm writing a protocol that has one step that requires the user to perform multiple interventions without intervening robot activity. I'd like to have three consecutive `protocol.pause()` steps, each describing one of the atomic steps for the user and requiring a "Resume" click as acknowledgement. 

Currently, though, these consecutive pauses just skip to the last one. 

# Steps to reproduce

On my machine, this code reproduces the behavior:

```{python}
from opentrons import protocol_api
from numpy import ceil

metadata = {'apiLevel': '2.5',
            'author': 'Jon Sanders'}

def run(protocol: protocol_api.ProtocolContext):

    protocol.pause('Click one')

    protocol.pause('Click two')

    protocol.pause('Click three')

```

# Current behavior

Only displays an interaction modal for the last pause command in a series. 

# Expected behavior

Should require interaction for each pause.

SyntaxColoring, 2020-08-12:
Sorry @tanaes, just saw this one. Thanks for reporting it. I've added this to our internal tracking for us to look into.

mcous, 2022-07-26:
Closing in favor of #8489

closed 2022-07-26 by mcous