Opentrons/opentrons #6214: feat: unload labware to reuse deck space in a protocol

rachelglover, 2020-07-24:
# Overview

I have some protocols that would really benefit from a protocol where the user can replace labware with something else. For example, part way through a protocol a tip rack will be empty. The protocol could be paused with an instruction for the user to remove the empty tip rack and replace with a 96-well PCR plate. This would enable more complex protocols which require a lot of tips or plates to re-use space on the deck without using multiple scripts.

# Implementation details

I'm thinking something along the lines of:
`protocol.pause("Remove the empty tip rack from position 7 and replace with a 96 well PCR plate")
**protocol.unload_labware('opentrons_opentrons_96_tiprack_300ul', 7)**
protocol.load_labware('biorad_96_wellplate_200ul_pcr',7)
`

theosanderson, 2020-07-25:
This took me ages to work out, so it would be very useful to document it, but you can delete the labware with:

```
rack= protocol.load_labware('tiprack',7) 
del protocol.deck['7']
plate = protocol.load_labware('biorad_96_wellplate_200ul_pcr',7) 
```

wolfrickw, 2020-07-25:
What about the calibration of the new labware?

theosanderson, 2020-07-27:
Labware calibration is persisted across protocols so you could create another protocol with the new (or old, not sure) labware for calibration. Honestly I have actually only tried this in Jupyter, but I don't see why it would be different.

Zeftow, 2020-08-03:
> This took me ages to work out, so it would be very useful to document it, but you can delete the labware with:
> 
> ```
> rack= protocol.load_labware('tiprack',7) 
> del protocol.deck['7']
> plate = protocol.load_labware('biorad_96_wellplate_200ul_pcr',7) 
> ```

Thanks a lot, that is so useful, I have been looking for it for a while now !

SyntaxColoring, 2024-02-21:
Extended support for moving labware around, including onto and off of the robot's deck, was added a little while ago in Python Protocol API v2.15 and robot software version 7.0.0. The Opentrons App accounts for this when doing Labware Position Check (formerly "labware calibration").

https://docs.opentrons.com/v2/moving_labware.html

Thank you all for this request!

closed 2024-02-21 by SyntaxColoring