Opentrons/opentrons #6234: refactor: The robot should send labware definitions to the runapp in RPC

sfoster1, 2020-07-29:
Right now, the rpc session specifies the load names of all labware in a protocol session.

If the runapp needs the labware definitions (for instance, to render the labware in the deckmap), it looks up the definitions based on the name of the labware.

However, this will fail in two cases:
1. The protocol on the robot was uploaded by a different runapp using custom labware that the connecting runapp does not have
2. The protocol on the robot is a JSON protocol that uses embedded custom labware that the run app does not have

In these cases, the run app cannot look up the labware definition.

To fix this, the robot should specify definitions for all the protocol's labware in the rpc, and the runapp should use those definitions in the protocol pages to render labware and do whatever else it needs definitions for.

amitlissack, 2021-01-28:
Does this relate to labware calibration? 

Since we're hoping to not use RPC for labware calibration, perhaps [ticket](https://github.com/Opentrons/opentrons/issues/7260 ) can supersede this one?

mattwelch, 2021-09-02:
This is not a thing anymore.

mattwelch, 2021-09-02:
We will not be sending LW defs via RPC. That's a PE/http thing now.

closed 2021-09-02 by mattwelch