Opentrons/opentrons #6238: bug: USB Connect Doesn't Work

gfleetwood, 2020-07-30:
## Overview

I'm on Ubuntu 18.04 using v3.19.0. I tried to enable ssh several times for the robot but the app would not recognize the usb connection - despite giving me a wired IP address - by:

* Failing to show the usb connection sign

* Failing to upload the ssh key using the [curl command in the docs](https://support.opentrons.com/en/articles/3203681-setting-up-ssh-access-to-your-ot-2). (It gave a port error.)

Trying to remedy this I switched off the wifi hypothesizing that it might be part of the interference, not remembering that the only way to turn it back on was by a usb connection. Now I have a non-functioning robot. 

I tried using three different A/B USB cables and even tried a BB USB cable connection straight to the raspberry pi, so I'd suspect that the cable isn't the issue. I also tried using a computer running Windows 10 instead (a dual boot with Ubuntu if it matters). 

What troubleshooting options do I have?

nickcrider, 2020-07-31:
Hi @gfleetwood, sorry you are seeing this. We do have some [basic info on troubleshooting connectivity issues here](https://support.opentrons.com/en/articles/2687601-troubleshooting-connection-problems), but it looks like you may have already seen that. It sounds like since you are getting an IP address on the USB interface this could be a problem with your network config causing the [mDNS (bonjuor)](https://en.wikipedia.org/wiki/Multicast_DNS) multicast discovery packets to get dropped/misrouted thus causing the app to not see the robot.

But unfortunately, I don't have enough info yet to make a complete diagnosis. Some additional questions for you:
- Have you ever been able to connect to this robot on your computer via the App?
- Other than disabling the wifi (ostensibly through the app?) and the failed `curl` command, what else has changed since you were able to connect to the robot previously?
- It's also worth double checking your network configuration to make sure [these ports are open](https://support.opentrons.com/en/articles/2687543-network-requirements-for-the-ot-2).
- Can you get an IP for the robot via `arp-scan` or similar? If avahi is configured correctly, you should also be able to ping the robot at `{robot S/N OT2CEP20...}.local`.
- The robot API HTTP server lives on port `31950` and once you have an IP for the robot you should be able to get a response at `{robot_ip}:31950/health`
- If you can discover the robot's IP and you do get a response from the health endpoint, you might be able to work around the multicast routing issue by using the field in the Network & System portion of the app to manually specify the robot's IP address.

Once I know from you a bit more about the nature of the problem, I should be able to restore connectivity to the robot. I am happy to support you through this channel, but if you'd like, you can get a faster, real-time chat response (during US business hours) on our website or by emailing support@opentrons.com too.

gfleetwood, 2020-08-03:
@nickcrider 

* Yes, I've connected with the app via wifi multiple times. However, the robot was set up a long time before I started using it, and I've never been able to connect via usb.
* Maybe a software update? I usually don't use the app. Most of my work with the robot is via scripts or the Jupyter notebook.

I'll have to wait until I'm with the robot again to try out the rest of your suggestions. Thanks!

mcous, 2020-08-03:
Hey @gfleetwood, I'm going to close this ticket since it's a support request. We'll re-open it if it leads to any identifiable software bug, and feel free to continue posting questions here and we'll do our best to answer them!

To add a couple things to what @nickcrider mentioned:

- With the OT-2 plugged in via USB, open the Opentrons App and navigate to More > Network & System > System Information
    - "USB-to-Ethernet Adapter Information" should show a connected device _and_ a **computer-side** link-local IPv4 address
    - If your computer does not have a link-local IP address, it will not be able to find the robot's IP address
- Two common causes for no computer-side IP address on this connection are:
    - Linux: the USB-to-Ethernet adapter is not configured to use link-local IP address on your computer
        - Our connectivity troubleshooting guide shows how to set this up on Ubuntu
    - Windows: your computer's USB-to-Ethernet driver is out of date
        - If your driver is out of date, the app will alert you and provide a link to update the driver
        - Windows Update is usually also able to update the driver depending

closed 2020-08-03 by mcous