Opentrons/opentrons #6311: bug: Pipette doesn't fully deliver the amount of liquid as requested.

ffffffrank, 2020-08-07:
# Overview

# Steps to reproduce

# Current behavior

# Expected behavior

MarcMarc94, 2020-08-12:
Which pipettes are you using? We had/have similar problems with the Gen1 pipettes. We replaced one of them and the problem was gone.

SyntaxColoring, 2020-08-12:
Hey @ffffffrank, thanks for opening this issue. As @MarcMarc94 mentioned, this sounds like it could be a problem with the pipette. (But without more details, I can't be sure.)

If you haven't already, please reach out to us at support@opentrons.com, and we'll work with you to troubleshoot.

closed 2020-08-12 by SyntaxColoring

ffffffrank, 2020-08-13:
> Which pipettes are you using? We had/have similar problems with the Gen1 pipettes. We replaced one of them and the problem was gone.

I have Gen1 pipette.