Opentrons/opentrons #6337: bug: When 2 pipettes are used, their calibration is not consistent, i.e. they don't move to the same positions.

MarcMarc94, 2020-08-12:
# Overview

When using two pipettes in one protocol, i.e. p10 (GEN1) and p50 (GEN1), their calibration is not consistent. When using a low volume 384 well plate and calibrating the labware with the p10, the p50 moves farther to the right than the p10 during aspiration/dispensing. This was already the case for API1 but seems much more exaggerated in API2. p10 can be used for the whole plate and the calibration is accurate enough. However, the p50 moves in between wells and is therefore not usable anymore.

# Steps to reproduce

1. Create a protocol with two pipettes and a 384 low volume plate (i.e. https://shop.gbo.com/en/england/products/bioscience/microplates/384-well-microplates/384-well-small-volume-lobase-microplates/788876.html). 
2. Calibrate the labware using the p10. 
3. Fill the plate with both pipettes.

# Current behavior

p10 works flawlessly for the whole plate. 
p50 is not working at all. It moves so far to the right of the expected point that it dispenses in between wells.

# Expected behavior

p50 should move to the exact same spot as p10 was calibrated.

mcous, 2020-08-12:
Hey @MarcMarc94, it sounds like there may be a problem with your robot's calibration data, specifically the piece of calibration data that tells the robot the relative positions of your pipettes. Have you tried the [Calibration Check tool](https://support.opentrons.com/en/articles/4028788-checking-your-ot-2-s-robot-calibration), and if so, what does it report?

MarcMarc94, 2020-08-13:
Thanks for the quick response. We will do that and let you know asap.

MarcMarc94, 2020-08-13:
I did the check, got an error that said the z axis was out of tolerance. I recalibrated the deck, ran the check again and z axis is still not in tolerance. x and y are fine. Which is a bit surprising because I don't have any problems with the z axis, only with x and y.
Here is the log file from the latest check (it didn't let me upload it as a file):

{"comparisonsByStep":{"comparingFirstPipetteHeight":{"differenceVector":[0,0,-3.6000000000000085],"thresholdVector":[0,0,0.8],"exceedsThreshold":true,"transformType":"BAD_DECK_TRANSFORM"},"comparingFirstPipettePointOne":{"differenceVector":[0.6999999999999975,0,0],"thresholdVector":[1.4,1.4,0],"exceedsThreshold":false,"transformType":"UNKNOWN"},"comparingFirstPipettePointTwo":{"differenceVector":[0.9000000000002046,-0.09999999999999964,0],"thresholdVector":[1.4,1.4,0],"exceedsThreshold":false,"transformType":"UNKNOWN"},"comparingFirstPipettePointThree":{"differenceVector":[0.9999999999999964,-0.20000000000004547,0],"thresholdVector":[1.4,1.4,0],"exceedsThreshold":false,"transformType":"UNKNOWN"},"comparingSecondPipetteHeight":{"differenceVector":[0,0,-1.70000000000001],"thresholdVector":[0,0,0.8],"exceedsThreshold":true,"transformType":"UNKNOWN"},"comparingSecondPipettePointOne":{"differenceVector":[-0.6000000000000085,0.3999999999999986,0],"thresholdVector":[1.8,1.8,0],"exceedsThreshold":false,"transformType":"UNKNOWN"}},"instrumentsByMount":{"LEFT":{"model":"p50_single_v1.3","name":"p50_single","tip_length":51.7,"mount":"LEFT","has_tip":true,"tiprack_id":"c4852357-69af-4d32-b560-b33d6e001b61","rank":"second","serial":"P50SV1318100940"},"RIGHT":{"model":"p10_single_v1.5","name":"p10_single","tip_length":33,"mount":"RIGHT","has_tip":false,"tiprack_id":"10e42489-cf72-4c37-9e9a-51207b71f7d6","rank":"first","serial":"P10SV1519112109"}},"calibrationStatus":{"deckCalibration":{"status":"OK","data":[[0.9987,0.0004,0,1.9064],[0.0008,1.0008,0,2.0801],[0,0,1,-24.1],[0,0,0,1]]},"instrumentCalibration":{"right":{"single":[-1.3761462589965276,0.33519883859167976,0],"multi":[0,0,0]},"left":{"single":[0.6994531447488725,1.1144766786383116,-0.6570000000000107],"multi":[0,0,0]}}},"savedAt":"2020-08-13T11:10:35.647Z"}

SyntaxColoring, 2020-08-13:
> I recalibrated the deck, ran the check again and z axis is still not in tolerance

Before you recalibrated the deck, did you factory-reset pipette calibration?

Bug #5022 (fixed in the yet-to-be-released v3.20.0) requires this, or the calibration will come out wrong. Our [Help Center](https://support.opentrons.com/en/articles/4028788-checking-your-ot-2-s-calibration) instructions account for this, but the in-app instructions don't.

MarcMarc94, 2020-08-14:
No, I didn't. Will try this next time. Thank you!

mcous, 2020-10-14:
Closing due to lack of activity and an assumption that this was a calibration data issue. @MarcMarc94 please feel free to chime back in if you're still experiencing this issue

closed 2020-10-14 by mcous