Opentrons/opentrons #6461: bug: App tells Intercom that unknown users are literally named "Unknown User"

SyntaxColoring, 2020-09-01:
# Overview

* Intercom has a few places that identify users by first name. ("Hi, Max!")
* It has smart fallbacks for when we don't know the user's name. ("Hi, there!")
* The app accidentally circumvents those fallbacks by setting the user's full name to "Unknown User":

   [image]

   [image]

# Steps to reproduce

1. Get a totally fresh Opentrons App installation. (You'll probably need to delete `Application Support/Opentrons`.)
2. Open a chat.
3. On Intercom, see that this is how you're identified:

[image]

# Current behavior

The app sends this to Intercom upon first contact:

https://github.com/Opentrons/opentrons/blob/1886fd172cfeb2a76e6a165b080c202a5c44cc8b/app-shell/src/config/migrate.js#L66

# Expected behavior

I *think* the app should send `name: null`, but there's also a chance it should send `name: ""`, or even not send a `name` at all. I haven't had a chance to look deeply into Intercom's user attribute semantics.

SyntaxColoring, 2020-09-01:
As a quick partial fix, I've changed the Intercom chat box welcome message to "Hi there! 👋", ignoring the user's name.

mcous, 2020-09-01:
This was an intentional decision to address the following:

- We assign a given app installation a unique ID to preserve identity across app openings (see #1999)
- In the absence of a known name or email, Intercom will display that user id in the inbound panel as the users name, which felt annoying at the time

[image]

The downsides of assigning the literal "Unknown User" to the name (e.g. having working fallbacks for an unknown name in automatic messages) seem to pretty clearly outweigh the downsides that lead us to implement `Unknown User` in the first place, but we should be aware that **removing the default name will cause UUIDs to be displayed at the users' names on our end**

PR #6468 (2020-09-02): fix(app): do not set the user's Intercom name to "Unknown User"
# Overview

This PR fixes #6461. See ticket for rationale and acceptance criteria 

# Changelog

- fix(app): do not set the user's Intercom name to "Unknown User"

# Review requests

- The app has never had logic to set `config.support.name` nor `config.support.email`, which is why I think its ok that we brute-force `null` them out rather than checking if the value matches the default before nulling it out
- **For support team members**: This will cause new users to show up in Intercom with their display name (not their actual nam [... cut]
commit 66662a49 (2020-09-02): fix(app): do not set the user's Intercom name to "Unknown User" (#6468)

Fixes #6461
closed 2020-09-02 by mcous