Opentrons/opentrons #6463: JSON V6 Schema + Executor

shlokamin, 2020-09-01:
# Overview

There is currently no way to communicate a `Dispense Air Gap` command to the API. In PD, we manually replace all `dispenseAirGap` commands with `dispense` commands on protocol export. We need a new executor + schema with a `dispenseAirGap` command.

# Implementation details

Add a dispense `dispenseAirGap` command to the new executor + schema. On the executor, it should to the same thing that `dispense` does.

# Acceptance criteria

Add the new command, review the [schema bump checklist](https://github.com/Opentrons/opentrons/pull/6285) and make sure that all criteria are fulfilled.

IanLondon, 2020-09-08:
Cross-linking: Related to #6399

mcous, 2021-06-11:
For the purposes of system alignment, we should also be discussing changes on both ends to match ProtocolEngine commands with JSON Protocol commands. Currently, there is a one-to-one mapping, so complete alignment should be technically feasible. In advance of a more full audit, currently implemented ProtocolEngine commands may be found here:

https://github.com/Opentrons/opentrons/tree/edge/api/src/opentrons/protocol_engine/commands

(As of the time of writing for this comment, the only thing to pay attention to is the payload shapes, currently named `SomeCommandRequest`). The shape of everything else is currently in flux as we simplify things to hook into the HTTP API)

IanLondon, 2021-06-11:
Duplicate of #6399

closed 2021-06-11 by IanLondon