Opentrons/opentrons #6466: feat: App: Option to disable the new app version popup when launching the application

egauthierphy, 2020-09-01:
# Overview

In a production environment the software versions are usually controlled and approved before being put in operation. The problem is that when starting the app, if a new software version is available a popup is displayed recommanding the user to upgrade the app and robot software. It is quite easy to download and install the new version. As a protection we can block the current Windows user from installing a new software, in which case the new version install will fail. However the popup will still re-appear at the next startup since it has not been upgraded.

An option in the app to disable this popup would be a great addition!

# Implementation details

Add an option to disable the new app version popup at startup.

mcous, 2020-09-01:
Thanks for the great request! This is on our radar

mcous, 2020-10-08:
Superseded by #6684 which is schedule for the 3.21.1 release. Please see that ticket for tracking!

closed 2020-10-08 by mcous