Opentrons/opentrons #6482: PD: Air gap on dispense

howisthisnamenottakenyet, 2020-09-03:
## Background
As a user I may want to air gap after a dispense to prevent drips/contamination between wells (in a multi dispense) and during pipette movement across the deck.

## Acceptance Criteria
- [ ] Add a checkbox for air gap after Blow out (https://www.figma.com/file/0dTEbGNsKYgNwKEW6LLfzJ/Airgap-Delay?node-id=50%3A850)
- [ ] Add tooltip on hover (https://www.figma.com/file/0dTEbGNsKYgNwKEW6LLfzJ/Airgap-Delay?node-id=104%3A0)
- [ ] **Min validation** Air gap default is the pipette's min. If there is not enough room left in the tip the default will be the largest value possible between the pipette's min and 0. If a user goes beneath the pipette's recommended minimum the user will get a dismissible warning.
- [ ] **Max validation** 
The remaining volume in a tip after the first (or only) dispense is as follows:
(Tip capacity - occupied volume of tip before the dispense) + first dispense volume + air gap volume.
If the user is doing a multi aspirate or 1:1 Transfer then the remaining volume in the entire tip capacity. The max is therefore the tip's capacity.
If a user is doing a multi dispense then the space for an airgap after the first dispense is any left over room in the tip after the aspirate, plus what was just dispensed (first dispense volume + air gap volume). This number gets bigger as the tip does more dispenses, but to keep things simple we will set the max at the smallest air gap.

shlokamin, 2020-09-08:
In terms of breaking this down into more technical, bite sized pieces, I see the tickets as:
 - Add checkbox + tooltip for air gap on dispense (UI) 
 - Add air gap dispense min validation for air gap on dispense (all three paths)
 - Add air gap dispense max validation for 1:1 transfer and multi aspirate
 - Add air gap dispense  max validation for multi dispense
 - Add atomic commands for air gap dispense for single transfer
 - Add atomic commands for air gap dispense for multi aspirate 
 - Add atomic commands for air gap dispense for multi dispense

shlokamin, 2020-09-10:
Closing this as it is broken down into:

- #6347
- #6500
- #6508
- #6510
- #6512

closed 2020-09-10 by shlokamin

shlokamin, 2020-09-10:
Note, still need a ticket for max air gap volume validation for multi dispense. Need to clarify whether it is okay if we clamp the maximum value down to whatever the max air gap aspirate value is. @howisthisnamenottakenyet