Opentrons/opentrons #6585: bug: App sometimes spins indefinitely waiting for an OT-2 to restart, because of a discovery client race condition

SyntaxColoring, 2020-09-21:
# Overview

When you update an OT-2, the Opentrons App shows a loading spinner while the OT-2 restarts.

Sometimes, that spinner never goes away. Even if the OT-2 has apparently restarted successfully. You need to close and reopen the Opentrons App to continue using the OT-2.

# Steps to reproduce

Upgrade a robot from v3.14.0 to v3.20.1. (This reliably triggered the problem on at least 10 robots, at one recent installation.)

# Current behavior

1. App asks update-server to restart by `POST /server/restart`, server responds quickly with an ”ok I’ll restart” message
2. App sets discovery client to fast poll and marks the robot as restart-pending
3. Eventually, the robot-server goes down due to restart
4. By luck, a discovery client poll happens while the robot-server is down, and the app marks the robot as restarting
5. Eventually, the robot-server comes back up
6. A discovery client poll happens, and the app notices that a robot marked as restarting is now healthy, marking it as up

If (4) doesn't happen—for example, if the robot restarts too quickly—then the app will think the robot never began the restart.

# Expected behavior

* The app should be able to eventually detect when an OT-2 has completed a restart, even in the face of an unreliable network or unexpected timing.
* In case the OT-2 never restarts, the spinner should also have a sensible timeout.

SyntaxColoring, 2020-09-21:
# Workaround

Close and reopen the Opentrons App.

SyntaxColoring, 2020-09-21:
# Possible solutions

A couple of possible directions for solutions are explored in Opentrons' internal chat.

mcous, 2020-10-27:
Fix:

- Add boot ID to the update server health endpoint
- Have app check for boot ID change as positive confirmation of restart
- Probably add a restart spinner timeout on the app side

PR #7589 (2021-04-14): feat(app): allow robot restarts to track boot ID and timeout
## Overview

This PR moves restart tracking logic from a smear across the `robot-admin` and `buildroot` modules into a single epic in the `robot-admin` module. The new epic:

- Keeps existing up/down tracking behavior (watch for robots to go down and come back up)
    - This logic has a known race condition that could result in a restart being "missed"
- Now also tracks [boot ID](https://github.com/Opentrons/opentrons/pull/7404) as an additional "successful restart" avenue
    - Because we're comparing boot IDs, the race condition ab [... cut]
commit 3b33102f (2021-04-14): feat(app): allow robot restarts to track boot ID and timeout (#7589)

Closes #6585
closed 2021-04-14 by mcous