Opentrons/opentrons #6819: Address Python protocol issue re: custom labware using multichannel

anthonywilleee, 2020-10-19:
This issue aims to take a holistic approach to #[5085](https://github.com/Opentrons/opentrons/issues/5085) - a bug where API v2 protocols can't access the second row of custom 384 well plates.

This is a Python protocol issue with custom labware.  We are looking to solve for all custom labware that might use a multi-channel pipette.
PD solves this same problem by using geometry -- if all 8 tips of a multi-channel will fit in a set of wells (1 tip per well, or the reservoir scenario where 1 well accommodates all 8 tips). It never looks at metadata.format

We could make Python take the same geometric approach.

Some open questions / next steps:
- Are there also issues in Python with multi-channel pipette access for 96 plates, or for reservoirs, from Labware Creator?
- Def want to loop HMG in bc we're changing how we interpret labware defs. I think we talked about deprecating use of metadata.format in labware defs before a little while back.
- Approach to this is unclear and needs research & coordination.

Related Tickets:
https://github.com/Opentrons/opentrons/issues/5085

anthonywilleee, 2020-10-19:
@IanLondon - my first ticket! 🎉  Basically just pulled your write up from Slack. Please add as you see fit. I'll keep this on a low burn for right now

sanni-t, 2024-08-13:
Duplicate/ outdated

closed 2024-08-13 by sanni-t