Opentrons/opentrons #6847: RFC: Robot Server API v2 breaking changes for v4.

amitlissack, 2020-10-22:
# Overview

We've been living with the new http for a few months. In that time we've developed opinions about what's working and what is not. As 4.0 approaches, we think it's a good opportunity to make a few changes. The motivation is simplification.

# Proposed changes

- Remove the type field  in request and responses. This is not used. It can be added in the future if we find a value for it.
- Remove the meta field in request and responses. Same rationale as type.
- Remove attributes . It is not necessary to force another level of nesting.

# Example request
```
{
	"data": {
		"type": "SessionCommand",
		"attributes": {
			"command": "equipment.loadLabware",
			"data": {
				"location": 2,
				"uri": "some",
				"namespace": "opentrons",
				"displayName": "mylabware",
				"version": 1
			}
		}
	}
}
```
becomes
```
{
	"data": {
		"command": "equipment.loadLabware",
		"location": 2,
		"uri": "some",
		"namespace": "opentrons",
		"displayName": "mylabware",
		"version": 1
	}
}
```

# Example response
```
{
  "links": {
    "self": {
      "href": "/sessions/5954e347-3e22-4189-bb5a-95e4041997a2"
    },
    "commandExecute": {
      "href": "/sessions/5954e347-3e22-4189-bb5a-95e4041997a2/commands/execute"
    },
    "sessions": {
      "href": "/sessions"
    },
    "sessionById": {
      "href": "/sessions/{sessionId}"
    }
  },
  "data": {
    "id": "5954e347-3e22-4189-bb5a-95e4041997a2",
    "type": "Session",
    "attributes": {
      "createParams": null,
      "sessionType": "liveProtocol",
      "details": {},
      "createdAt": "2020-10-22T20:07:57.717947+00:00"
    }
  }
}
```
becomes
```
{
  "links": {
    "self": {
      "href": "/sessions/5954e347-3e22-4189-bb5a-95e4041997a2"
    },
    "commandExecute": {
      "href": "/sessions/5954e347-3e22-4189-bb5a-95e4041997a2/commands/execute"
    },
    "sessions": {
      "href": "/sessions"
    },
    "sessionById": {
      "href": "/sessions/{sessionId}"
    }
  },
  "data": {
    "id": "5954e347-3e22-4189-bb5a-95e4041997a2",
    "createParams": null,
    "sessionType": "liveProtocol",
    "details": {},
    "createdAt": "2020-10-22T20:07:57.717947+00:00"
  }
}
```

sfoster1, 2020-10-22:
Big fan! Gonna be a big change set though

amitlissack, 2020-10-23:
There are no obejctions so we will go forward with this.

mcous, 2020-11-05:
RFC accepted, moved to epic #6848

closed 2020-11-05 by mcous