Opentrons/opentrons #6910: bug: Labware Creator and Opentrons App accept bad labware definitions in the X-axis

alexjoel42, 2020-11-04:
# Overview

Too large X Well Spacing and X Grid Offset values are able to be input into both Labware Creator and the Opentrons App

# Steps to reproduce

- Open Labware Creator 
- Input a large well spacing and/or grid offset value 
[image]

- Download the labware definition
[Bad_x_grid_and_spacing.zip](https://github.com/Opentrons/opentrons/files/5488979/Bad_x_grid_and_spacing.zip)

- Upload the definition to either Labware Creator or Opentrons App 

# Current behavior

The definition gets uploaded without issue. 
# Expected behavior

When the y grid offset or well spacing is modified to be similarly large, the following error appears: 

Labware Creator:

[image]. 

Opentrons App: 
[image]

The boundary conditions should behave the same in both the x and y axis when considering grid offset and well spacing.

anthonywilleee, 2020-11-09:
need to start with validation in labware creator

IanLondon, 2020-12-11:
Hmm... I can't replicate this validation error in Run App with the attached labware def or labware test .py protocol. Also not sure how to make negative coordinates show up in Labware Creator's output, I can make wells that far fall off the plate in positive x/y directions, but not down the negative axes.

anthonywilleee, 2021-06-22:
This will be fixed for labware creator with custom tip rack work, but error will not be displayed on Run app. Once we release LC, this will no longer be a problem, but if its a definition that was created prior to custom tipracks LC update, this will still be an issue in Run App.

Consider breaking out separate bug / ticket for Run app

IanLondon, 2021-06-22:
^ Just to link tickets: addressed for LC in #7165 (but not addressed in Run App)

IanLondon, 2021-07-16:
Gonna close this bc after #7165 it's a Run App thing, not an LC thing

closed 2021-07-16 by IanLondon