Opentrons/opentrons #6951: bug: GET /settings/pipettes takes a long time and blocks other requests

SyntaxColoring, 2020-11-10:
# Overview / current behavior

Hitting the robot's `GET /settings/pipettes` endpoint:

* Takes a long time. (~4 seconds on my OT-2, I think scaling with the number of pipettes for which the OT-2 is storing settings.)
* Synchronously blocks subsequent requests.

One user-facing effect of this is that the calibration jog buttons will have a latency of up to ~4 seconds. The app polls `GET /settings/pipettes` every 5 seconds during pipette offset calibration, so a jog action will usually get stuck waiting for that poll to complete.

# Steps to reproduce

1. Transplant [my OT-2's `/data/pipettes` directory](https://github.com/Opentrons/opentrons/files/5513938/pipettes.zip) onto your OT-2.

2. As a benchmark, run:

    ```
    curl --location --request GET '<robot_ip>:31950/modules' --header 'Opentrons-Version: 2'
    ```
    
    And see that it returns after a fraction of a second, as expected.

3. Run the problematic command:
    
    ```
    curl --location --request GET `<robot_ip>:31950/settings/pipettes' --header 'Opentrons-Version: 2'
    ```
    
    And see that it returns after approximately 4 seconds, a long time.

4. Run the command from step 3 and then immediately run the command from step 2. See that they *both* return after approximately 4 seconds; the long command blocked the short command.

Tested on robot and app 4.0.0-alpha.7.

# Expected behavior

* While `GET /settings/pipettes` is doing its thing, a concurrent unrelated request, like `GET /modules`, should be processed and responded to quickly.
* `GET /settings/pipettes` should probably not take as long as 4.2 seconds? (Not sure. My OT-2 has seen an unusually high number of pipettes. The duration may be suitably short on OT-2s that have seen fewer pipettes.)

amitlissack, 2020-11-10:
@SyntaxColoring what are the contents of the `.opentrons/pipettes` directory on your robot?

amitlissack, 2020-11-10:
The handler for `/settings/pipette` calls `pipette_config.known_pipettes` walks the pipette config directory finding configurations.
After that, for each pipette file we call `pipette_config.list_mutable_configs`. `list_mutable_configs` in turn calls `pipette_config.known_pipettes` to get all the pipette ids. It then does linear search in the `known_pipettes` list. 
suggestions:
- `list_mutable_configs` accepts `known_pipettes` as input.
- `known_pipettes` returns a Set or `list_mutable_configs` creates a Set from input list.

sfoster1, 2020-11-10:
I poked at it with a "quick fix" of changing `list_mutable_configs` to look like this:
```
def list_mutable_configs(pipette_id: str) -> Dict[str, Any]:
    """
    Returns dict of mutable configs only.
    """
    cfg: Dict[str, Any] = {}

    try:
        config, model = load_config_dict(pipette_id)
    except FileNotFoundError:
        log.info(f'Pipette id {pipette_id} not found')
        return cfg
    ...
```

which is a) more pythonic and b) avoids the quadratic issue. it's still pretty slow! and the reason appears to be that while the code is complex, at its base `opentrons_shared_data.pipette.fuse_specs` takes like 40ms because parsing the stupid gigantic pipette model spec is hard. tossing that in a cache fixes that issue and cuts the time if you have the attached pipettes dict down. Doing those two things cuts the time for the request to about a second (after the cache is built) but that's still not good!

Matt-Zwimpfer, 2021-06-03:
MLM - not immediate fix; check with @sfoster1 on the "quick fix" status of this

SyntaxColoring, 2025-12-15:
Stale.

closed 2025-12-15 by SyntaxColoring