Opentrons/opentrons #7048: bug: Version 4, adding attributes to labware is not working anymore

MarcMarc94, 2020-11-24:
# Overview

In our protocol we're assigning properties to our labware. This used to work without any problems. Now however it doesn't anymore. Every time we try to add a new property to a labware or a well, it seems to create a copy instead of adding it to the original labware. Example code:
 
for w in mainprotocol.loaded_labwares[slot].wells():
       w.properties={}
 
if we follow this by raising an Exception:
 
raise Exception(str(dir(mainprotocol.loaded_labwares[slot])), str(dir(mainprotocol.loaded_labwares[slot]['A1'])), str(dir(w)), str(hasattr(mainprotocol.loaded_labwares[slot]['A1'], 'get_name')))
 
w does have the attribute properties, the original labware however, does not.
 
It doesn't even work if we do it explicitly, i.e. mainprotocol.loaded_labwares[slot]['A1'].properties = {}

Another problem is that the opentrons library doesn't seem to be up to date. Therefore we have to try to debug it on the robot - that's why we use raise Exception.

mcous, 2020-11-24:
@MarcMarc94 thanks for the report! It looks like our deploy server didn't release 4.0.0 to PyPI properly; I'll get that sorted out right now.

As for your other question:

> In our protocol we're assigning properties to our labware. This used to work without any problems. Now however it doesn't anymore.

This is an expected consequence of #6716. Behavior changes like this are why we classified the release as 4.0.0 instead of 3.x. In general moving forward, **you should not mutate custom and/or undocumented properties of Protocol API objects.** Except for documented setter interfaces, you should treat all objects returned to you as read-only to avoid future breaks.

If you need to "attach" data to a well or other object, I recommend that you set up a separate data storage object and get/set your custom data there, by reference:

```python
# ...
well_properties = {}

for w in mainprotocol.loaded_labwares[slot].wells():
    well_properties[w] = {}

# ...later
some_well = mainprotocol.loaded_labwares[slot]['A1']
some_well_properties = well_properties[some_well]
```

mcous, 2020-11-24:
v4 has been deployed to PyPI and we're tracking the server misconfiguration that caused the failed deploy in the ticket shown above. I'm going to close this report as expected behavior of the Protocol API, and will follow up with our support and apps engineering teams to make sure we don't recommend mutating protocol objects to users

closed 2020-11-24 by mcous