Opentrons/opentrons #7051: RFC: switch Python docstring convention to Google-style

mcous, 2020-11-24:
## overview

We currently specify our docstrings in the default, Sphinx format:

```python
    @requires_version(2, 0)
    def load_labware(
            self,
            load_name: str,
            location: types.DeckLocation,
            label: str = None,
            namespace: str = None,
            version: int = None,
    ) -> Labware:
        """ Load a labware onto the deck given its name.

        For labware already defined by Opentrons, this is a convenient way
        # ...etc...

        :param load_name: A string to use for looking up a labware definition
        :param location: The slot into which to load the labware such as
                         1 or '1'
        :type location: int or str
        :param str label: An optional special name to give the labware. If
                          specified, this is the name the labware will appear
                          as in the run log and the calibration view in the
                          Opentrons app.
        :param str namespace: The namespace the labware definition belongs to.
            If unspecified, will search 'opentrons' then 'custom_beta'
        :param int version: The version of the labware definition. If
            unspecified, will use version 1.
        """
```

(If you go into some of our legacy code, you can also find NumPy-style comments... but let's ignore that).

Speaking for myself, I find these `:param`, `:type`, etc. directives a little difficult to write and even more difficult to read. **I would like to propose that we move over to [Google-style docstrings](https://google.github.io/styleguide/pyguide.html#38-comments-and-docstrings)** to make these friendlier to write and read:

```python
    @requires_version(2, 0)
    def load_labware(
            self,
            load_name: str,
            location: types.DeckLocation,
            label: str = None,
            namespace: str = None,
            version: int = None,
    ) -> Labware:
        """Load a labware onto the deck given its name.

        For labware already defined by Opentrons, this is a convenient way
        # ...etc...

        Parameters:
            load_name: A string to use for looking up a labware definition.
            location: The slot into which to load the labware such as 1 or '1'.
            label: An optional special name to give the labware. If
                specified, this is the name the labware will appear
                as in the run log and the calibration view in the
                Opentrons app.
            namespace: The namespace the labware definition belongs to.
                If unspecified, will search 'opentrons' then 'custom_beta'.
            version: The version of the labware definition. If
                unspecified, will use version 1.

        Returns:
            The labware instance.
        """

## why?

My opinions on this matter:

- They're easier to read, with less cruft
- Separate sections for parameters, returns, examples are really nice, and this would enforce them
- Types of parameters are read from type annotations
- There's a larger ecosystem of doc generation tools that _aren't_ sphinx that are compatible with Google-style docstrings
    - Specifically, I'm thinking of https://github.com/pawamoy/mkdocstrings/
- pydocstyle can check for missing parameter descriptions in Google-style
    - Note: It can get tripped up pretty easily and fail to complain, but it's [actively being worked on](https://github.com/PyCQA/pydocstyle/milestone/6)

## how?

1. Switch the flake8-docstrings linting config from "pep257" to "google"
2. Ensure [sphinx.ext.napoleon](https://www.sphinx-doc.org/en/master/usage/extensions/napoleon.html) (already included in our Sphinx config) is properly configured for type annotations, etc.

SyntaxColoring, 2020-11-24:
Note that switching to Google docstring syntax won't actually let us eliminate arcane Sphinx (reStructuredText) syntax in docstrings.

For example, we'll still need it for:

* Cross-references.
    ```restructuredtext
    If you pass ``new_tip='never'``, there must already be a tip on the pipette,
    such as from :py:obj:`pick_up_tip`.
    ```
* Warning blocks.
    ```restructuredtext
    .. warning::
        Don't set ``home_after=False`` unless you know what you're doing.
    ```

So I worry that instead of having to learn one way of doing things, we'd have to learn two—and how they interact with each other.

But I agree with all your "Why"s, so I'm willing to give it a shot.

mcous, 2020-11-24:
@SyntaxColoring I think with warning blocks (and most other blocks) we'd be [covered by Napoleon](https://www.sphinx-doc.org/en/master/usage/extensions/napoleon.html#docstring-sections):

> Docstring Sections
> 
> All of the following section headers are supported:
> 
> - Args (alias of Parameters)
> - Arguments (alias of Parameters)
> - Attention
> - Attributes
> - Caution
> - Danger
> - Error
> - Example
> - Examples
> - Hint
> - Important
> - Keyword Args (alias of Keyword Arguments)
> - Keyword Arguments
> - Methods
> - Note
> - Notes
> - Other Parameters
> - Parameters
> - Return (alias of Returns)
> - Returns
> - Raise (alias of Raises)
> - Raises
> - References
> - See Also
> - Tip
> - Todo
> - Warning
> - Warnings (alias of Warning)
> - Warn (alias of Warns)
> - Warns
> - Yield (alias of Yields)
> - Yields

I think you're right about cross-references. My not-so-secret-hope would be that we could figure out a way to end up using Markdown-style cross-references in a manner similar to [mkdocstrings](https://pawamoy.github.io/mkdocstrings/usage/#cross-references), perhaps via a Sphinx plugin or via switching off of Sphinx

PR #7623 (2021-04-13): style(python): use flake8 for all Python linting
## Overview

This PR follows up on the work done in #6945 and switches remaining Python projects to [flake8](https://flake8.pycqa.org/en/latest/), for consistency across all Opentrons' Python projects.

I also took the opportunity to set `flake8-annotations` to Google-style, which closes #7051.

(As we're moving into more Protocol Engine wire-up work, the linter discrepancy was getting in the way of ensuring work felt properly greenfield)

## Changelog

- Switch `robot-server` to flake8
- Switch `update-server` to flake8
- Switch `shared-data/py [... cut]
commit 3bad6db6 (2021-04-13): style(python): use flake8 for all Python linting (#7623)

Closes #7051
closed 2021-04-13 by mcous