Opentrons/opentrons #7112: bug: Shouldn't you do `pipenv install python 3.7` in Makefile?

MarcelRobitaille, 2020-12-08:
# Overview

I have been trying to `make setup` for a while now, and the problem was that I have Python 3.9 on my system. Isn't the point of `pipenv` to give everyone the same environment? Therefore, shouldn't part of the `make` process be to install the required python version, 3.7?

I don't actually use `pipenv`, I prefer the simpler `python -m venv`, so I could be wrong.

# Steps to reproduce

1. Install Python 3.9 on your system (eg. `pacman -S python`)
1. Clone this repo
1. `make setup`

# Current behavior

1. pipenv tries to install python 3.7 packages using python 3.9
1. Install fails

# Expected behavior

1. pipenv first installs python 3.7 and continues with that python version
1. Install succeeds

mcous, 2020-12-09:
Hey @MarcelRobitaille, as detailed in out [contributing guide](https://github.com/Opentrons/opentrons/blob/edge/CONTRIBUTING.md#development-setup), you need to ensure Python 3.7 is installed on your system _prior_ to using pipenv. Pipenv is (usually) capable of finding Python 3.7 on your system to create its virtual environment, but it is not capable of installing Python itself.

We generally feel like trying to assert a particular Python installation setup via the Makefiles would be too much of a burden for contributors. Most of us on *nix use [pyenv](https://github.com/pyenv/pyenv) to manage our Python versions. It looks like the Arch wiki (assuming you're on Arch because of pacman, sorry if this is incorrect) [also recommends pyenv](https://wiki.archlinux.org/index.php/Python#Python_3).

So, first time setup looks something like (typing from memory here, this may not be 100% correct):

```shell
# install Python 3.7
pyenv install 3.7.9

# set Python 3.7 as your Python version
pyenv global 3.7.9

# run make setup, which will install pipenv et. al
# if make setup fails after installing pipenv, you may
# need to run `pyenv rehash`
make setup
```

closed 2020-12-09 by mcous

MarcelRobitaille, 2020-12-09:
Hi @mcous. Sorry for opening an issue about something that's documented. Thank you for the very detailed reply. Maybe a comment in [code blocks like this](https://github.com/Opentrons/opentrons/tree/edge/app#developing) could help?
```
# install Python 3.7 (pyenv install 3.7)
...
```