Opentrons/opentrons #7133: bug: Can't callibrate a left mounted p1000 gen2 with version 4.0.0

egenomics, 2020-12-15:
# Overview

When I click pippette offset callibration, then start tip length callibration, then set the two modules required (p1000 tiprack and metal module to callibrate).
Next step is to click confirm placement and continue. In this point the robot does nothing and just shows me the message again.

# Steps to reproduce

SO: MacOS
App and robot version: 4.0.0

# Current behavior

Can't callibrate pipette

# Expected behavior

Be able to callibrate pipette

jeppefock, 2020-12-22:
The same behavior is observed with a p20 pipette in the left mount

mcous, 2021-12-13:
Closing ticket due to age. If this issue has not been resolved, please do file a new ticket!

closed 2021-12-13 by mcous