Opentrons/opentrons #7134: bug: Where do I put custom labware?

MarcelRobitaille, 2020-12-15:
Not so much a bug as a question: Where am I supposed to put custom labware? The GUI wants it in `~/.config/Opentrons/labware/custom_10_tuberack_10x15ml.json` but the simulator wants it in `~/.opentrons/labware/v2/custom_definitions/quin/custom_10_tuberack_10x15ml/1.json`. 

I have read through every piece of documentation I could find on custom labware, but none seem to specify where to actually *put* the labware files: 
https://github.com/Opentrons/opentrons/tree/edge/labware-designer
https://support.opentrons.com/en/articles/3136504-creating-custom-labware-definitions
https://support.opentrons.com/en/articles/3136501-what-is-a-labware-definition
https://support.opentrons.com/en/articles/1820316-creating-custom-labware-definitions-using-python-protocol-apiv1
https://docs.opentrons.com/v2/new_protocol_api.html#module-opentrons.protocol_api.labware
https://support.opentrons.com/en/articles/2687694-get-started-python-api-onboarding-protocol-tutorial

mcous, 2020-12-15:
Hi @MarcelRobitaille, thanks for your question. I agree that this is a frustrating part of the Opentrons system. To get you going, though:

- For Protocol Designer protocols, all you need to do is upload the custom labware definitions via Protocol Designer and you are set
- For Python protocols, add them via the Opentrons App (to the default location or whatever location you have selected via the GUI)
- To simulate Python protocols with custom labware, use the `-L, --custom-labware-path` option of `opentrons_simulate` and enter the custom labware directory your app is set up to use
    - See `opentrons_simulate --help` for more details

If you add custom labware to the app, you _must_ upload your Python protocol via the app, too. This is due to a pretty annoying limitation in robot's current protocol upload system. We're replacing this system over the next few releases, but if that limitation is a problem now, I would recommend reaching out via support chat for a different solution (feel free to reference this GH thread so I can fill in the support team).

Keeping this ticket open as a "documentation and experience needs improvement" issue

MarcelRobitaille, 2020-12-16:
Thank you for your detailed answer. 

I am still getting the same error with `-L ~/.config/Opentrons/labware`. Is it still looking for `~/.config/Opentrons/labware/namespace/name/version.json`? Right now, my `~/.config/Opentrons/labware` has no subfolders.

I understand that this is a big project and that good documentation is hard. Maybe for the short term, the simulate command could give some extra info in the error message?
Maybe instead of:
```
opentrons.protocols.execution.errors.ExceptionInProtocolError: FileNotFoundError [line 23]: Labware "custom_15ml_vial_rack" not found with version 1 in namespace "quin"
```
something like this:
```
Could not find "/path/to/where/the/file/is/expected". Maybe you need to use the -L flag?
```

MarcelRobitaille, 2020-12-16:
Still no luck. I have a copy of my labware definition in `~/.config/Opentrons/labware/custom_15ml_vial_rack.json` and another in `~/.config/Opentrons/labware/quin/custom_15ml_vial_rack/1.json`.

The only error I am getting is the following, suggesting that it is still only looking in `~/.opentrons...`.
FileNotFoundError: [Errno 2] No such file or directory: '/home/marcel/.opentrons/labware/v2/custom_definitions/quin/custom_15ml_vial_rack/1.json'

Matt-Zwimpfer, 2021-06-03:
MLM - not an immediate fix

GimenezEGT, 2023-08-04:
Hi, there. I'm having the same problem.

input: 
opentrons_simulate.exe Extracao_Bio_Gene_96.py --custom-labware-path C:\Users\enrico.gimenez\AppData\Roaming\Opentrons\labware
output:
FileNotFoundError [line 40]: Labware "CUSTOM LABWARE NAME" not found with version 1 in namespace "separation plate".

Is there any recommendation for this? I need to test it in siilco before running on equipament.

Thank you.

SyntaxColoring, 2024-02-21:
Closing:

* The [original question](https://github.com/Opentrons/opentrons/issues/7134#issue-767830760) of "where do I put custom labware" has been [answered](https://github.com/Opentrons/opentrons/issues/7134#issuecomment-745547518), and this is documented in `opentrons_simulate --help`.
* I'm not sure why that [wasn't working](https://github.com/Opentrons/opentrons/issues/7134#issuecomment-745750364) for @MarcelRobitaille, but I've just tested it and it appears to be working now. It's been a while, so there definitely might have been a bug that got fixed at some point.
* I suspect it [wasn't working](https://github.com/Opentrons/opentrons/issues/7134#issuecomment-1666127281) for @GimenezEGT because of a mistake in their `load_labware()` syntax. Their error message says `namespace "separation plate"`, and I suspect "separation plate" was meant to be provided for the `label` argument, not the `namespace` argument.

closed 2024-02-21 by SyntaxColoring