Opentrons/opentrons #7156: bug(api): Two pipettes in use, dispensing in same well; second pipette dispensing moves diagonally downwards.

Matt-Zwimpfer, 2020-12-22:
# Overview

# Steps to reproduce

# Current behavior

# Expected behavior

mcous, 2020-12-22:
Shoot, theorized this bug a few weeks back from reading the code but never got around to testing it. Glad to have the repro.

Using a pipette sets the ProtocolContext’s location cache, but changing the active pipette does not clear the cache, so the system thinks pipette number 2 is already inside the target well and it issues a direct movement rather than an arc.

Issue has already been accounted for and avoided in the newer ProtocolEngine, but should probably be fixed in the currently thick protocol API, too.

beniroquai, 2021-03-31:
Hey, is there any update on this? I face the same issue when performing the following:

```py
    for i_col in range(i_experiment, N_cols):
        for i_steps in range(2):
            pipette.dispense()  
            #print("Remove liquid "+experiment_methods[i_row_bluedye]+" - column: "+str(i_col))

            '''
            Go to the microscope and dispense 
            '''    
            #get position of the first well of the microscope
            center_location = plate.columns()[i_row][i_col].center()
            
            # offset-coordinates (only calibrate it once)
            # dx=-6; dy=2; dz=150

            # go to well
            adjusted_location = center_location.move(types.Point(x=-6, y=2, z=135))
            print(adjusted_location.point)
            pipette.move_to((adjusted_location))
            time.sleep(.51)
            
            # go to down and aspirate 
            adjusted_location = center_location.move(types.Point(x=-6, y=2, z=88))
            print(adjusted_location.point)
            pipette.move_to((adjusted_location))
            pipette.aspirate(V_aspirate_blue,adjusted_location)
            time.sleep(0.51)
            
            # move up
            adjusted_location = center_location.move(types.Point(x=-6, y=2, z=135))
            print(adjusted_location.point)
            pipette.move_to((adjusted_location))
            time.sleep(.51)

            '''
            Go to the liquid and aspirate some stuff
            '''
            #get position of the first well of the microscope
            pos_reagent = plate2.columns(1)[0][1].center()
            dx=0; dy=-3; dz=-15 # offset-coordinates (only calibrate it once)

            # Get a location 1 mm right, 1 mm back, and 1 mm up from the center of well A1.
            adjusted_location = pos_reagent.move(types.Point(x=dx, y=dy, z=dz))
            print(adjusted_location.point)
            pipette.move_to((adjusted_location))
            pipette.dispense()
            pipette.blow_out()

            # move head up
            # Get a location 1 mm right, 1 mm back, and 1 mm up from the center of well A1.
            adjusted_location = pos_reagent.move(types.Point(x=dx, y=dy, z=130))
            print(adjusted_location.point)
            pipette.move_to((adjusted_location))
```

where the aspirate causes this diagonal motion. A second pipette is mounted too but not yet moved. 

Thanks!

beniroquai, 2021-03-31:
@mcous 

interstingly, this diagonal motion is not hapenning in the second loop here:

```py

protocol.home()
i_pipette = pickup_fresh_pipette_tip(i_pipette, pipette, tiprack)
for i_steps in range(2):
            pipette.dispense()  
            #print("Remove liquid "+experiment_methods[i_row_bluedye]+" - column: "+str(i_col))

            '''
            Go to the microscope and dispense 
            '''    
            #get position of the first well of the microscope
            center_location = plate.columns()[i_row][i_col].center()

            # go to well
            adjusted_location = plate.columns()[i_row][i_col].center().move(types.Point(x=-6, y=2, z=135))
            print(adjusted_location.point)
            pipette.move_to((adjusted_location))
            
            # go to down and aspirate 
            adjusted_location = plate.columns()[i_row][i_col].center().move(types.Point(x=-6, y=2, z=88))
            print(adjusted_location.point)
            #pipette.move_to((adjusted_location))
            pipette.aspirate(50,adjusted_location)
            
            # move up
            adjusted_location = plate.columns()[i_row][i_col].center().move(types.Point(x=-6, y=2, z=135))
            print(adjusted_location.point)
            pipette.move_to((adjusted_location))

            '''
            Go to the liquid and aspirate some stuff
            '''
            #get position of the first well of the microscope
            pos_reagent = plate2.columns(1)[0][1].center()
            dx=0; dy=-3; dz=-15 # offset-coordinates (only calibrate it once)

            # Get a location 1 mm right, 1 mm back, and 1 mm up from the center of well A1.
            adjusted_location = pos_reagent.move(types.Point(x=dx, y=dy, z=dz))
            print(adjusted_location.point)
            pipette.move_to((adjusted_location))
            pipette.dispense()
            pipette.blow_out()

            # move head up
            # Get a location 1 mm right, 1 mm back, and 1 mm up from the center of well A1.
            adjusted_location = pos_reagent.move(types.Point(x=dx, y=dy, z=130))
            print(adjusted_location.point)
            pipette.move_to((adjusted_location))
```

Is there a quick fix for it?

PR #7609 (2021-04-14): fix(api): api location cache fix
# Overview

The location cache is used regardless of which pipette is moving. If left moves immediately after right, the left move's cached location will be used. This can cause jerky motion.

closes #7156 

# Changelog
- update protocol api version to 2.10
- maintain the last mount moved.

# Review requests

Test on a robot using example protocols in #7156 

# Risk assessment

High.
commit df68ea2d (2021-04-14): fix(api): api location cache fix (#7609)

* wip: noodle around with location cache
* add unit tests

closes #7156
closed 2021-04-14 by amitlissack