Opentrons/opentrons #7163: LC: New footprint error

howisthisnamenottakenyet, 2020-12-22:
## Background
As a user I would like to know that there is a way to get adapters for labware that is too small to fit in slots

## Design
[image]

## Acceptance criteria
- [ ] When a labware is too small (beyond the lower bounds of what is acceptable to the form field) user should see an additional error 
- [ ] The link in the error goes to: https://docs.google.com/forms/d/e/1FAIpQLScvsHlXQrtIhIQYO0zr6mYwmzOCGpYPqepeDIorFIyj2jT-UQ/viewform

## Remaining questions
@anthonywilleee 
1. I think there's a margin of error around when we show whether a labware is too big for a slot. We should find out if there's a margin of error for when a labware is too small.
2. We discussed doing this error for tip racks but why shouldn't we do it for all labware types?

anthonywilleee, 2020-12-22:
@howisthisnamenottakenyet 
1. agree, i can follow up with apps eng, unless you think otherwise. do you know what's the margin of error for too big?
2. can't think of a reason to not do this for all labware types, but i can add that q to apps eng

Kadee80, 2021-05-19:
MIN_X_DIMENSION = 127 MIN_Y_DIMENSION = 85

PR #7846 (2021-06-07): feat(labware-creator): Add new errors for too large/small footprint
# Overview

closes #7163 closes #7792 by adding the too big and too small footprint errors 

# Changelog

- feat(labware-creator): Add new errors for too large/small footprint

# Review requests

- [ ] Labware too small error renders when footprint X or Y is below minimum
- [ ] Labware too large error renders when footprint X or Y is above max
- [ ] Links open the appropriate forms

FYI:
```
MAX_X_DIMENSION = 129
MIN_X_DIMENSION = 127

MAX_Y_DIMENSION = 87
MIN_Y_DIMENSION = 85
```
# Risk assessment
Low, new errors in LC
commit dd853def (2021-06-07): feat(labware-creator): Add new errors for too large/small footprint (#7846)

closes #7163 closes #7792
closed 2021-06-07 by Kadee80