Opentrons/opentrons #7167: bug: Infrequent systematic issues with pause and run clock features

ethanfjones, 2020-12-24:
# Overview

There are three different events that seem to be happening with different OT-2s. 

**Event 1** 
When they hit pause ~30% it stops and then when they resume, automatically resumes to the next step and then pauses again. They need to spam the pause/resume button in order to get it to work ( I know not good)

[Event 1 .zip](https://github.com/Opentrons/opentrons/files/5739941/Event.1.zip)

**Event 2** 
The OT-2 was paused and the customer went for lunch. When they had gotten back to the OT-2 it said: "run complete". The time though was showing the time they hit the pause button at. (13 mins) According to the logs, the protocol ran perfectly fine.

[Event 2.zip](https://github.com/Opentrons/opentrons/files/5739942/Event.2.zip)

**Event 3**
The protocol is running but the timer is not starting on the run. This event seems to happen more when a protocol is restarted vs on a fresh protocol upload. 

[Event 3.zip](https://github.com/Opentrons/opentrons/files/5739943/Event.3.zip)

# Steps to reproduce

I have not tried to reproduce the issue yet

amitlissack, 2021-01-28:
We believe that Event 1 is fixed (#7216)

SyntaxColoring, 2021-01-29:
Closing this in favor of several smaller tickets.

closed 2021-01-29 by SyntaxColoring