Opentrons/opentrons #7179: bug: protocol.set_rail_lights() is not working from OT2 Jupyter notebook

kasuskasus1, 2021-01-06:
# Overview

protocol.set_rail_lights() is not working from OT2 Jupyter notebook

# Steps to reproduce

* Create Jupyter network at <ROBOT_IP>:48888
* Add the following code:
```
import opentrons.execute
from time import sleep
protocol = opentrons.execute.get_protocol_api('2.8')

def run(protocol):
    protocol.set_rail_lights(True)
    print(protocol.rail_lights_on)
    sleep(5)
    protocol.set_rail_lights(False)
    print(protocol.rail_lights_on)

run(protocol)
```

# Current behavior
Produced output:
```
True
```
after 5 sec:
```
True
```
Lights are not turning on and then off.

# Expected behavior
Expect lights to turn on and then turn off after 5 sec.

chazmanc, 2021-01-21:
@kasuskasus1  instead of using `pause(5)`, you can use `protocol.delay(seconds=5)` - this should give you the expected behavior in the meantime

alexjoel42, 2021-01-21:
Hi there! 
Thank you for your patience on this one. 

In appreciation of your patience, I have made a sample protocol that should help you with this task: 

Note the OT-2's circuitboard that controls the lights, rail lights, etc. can only take one connection and has to chose between the App and Jupyter Notebook. 

That is why we need to run: 
import os 
os.system("systemctl stop opentrons-robot-server")
in the protocol to make sure that the OT-2 only recognizes the Jupyter Notebook Connection (Theo makes a good point of this here https://github.com/Opentrons/opentrons/issues/6164#issuecomment-666506963 ) 

My recommendation is to divide the protocol into three cells: 

1. Stoping the robot server so Jupyter Notebook can control lights
2. controlling lights for a protocol
3. restarting the server so that you can use the app

## Cell One, before you manipulate lights
import opentrons.execute
import os

os.system("systemctl stop opentrons-robot-server")
`
## Cell Two Manipulating Lights 

import opentrons.execute
import os

protocol = opentrons.execute.get_protocol_api('2.8') 
protocol.home()
protocol.set_rail_lights(False)
protocol.set_rail_lights(True)
protocol.comment('light is on')
protocol.delay(seconds = 20)
protocol.set_rail_lights(False)
protocol.comment('light is off')
#resetting the server 
for line in protocol.commands():
    print(line)
'

##Cell Three Restarting the Server so that the App can connect

os.system("systemctl start opentrons-robot-server")

'

## For Reference

[image]

Here are some resources to make your own code based on the example above. 

1. Our Jupyter Notebook code [is a bit differently organized](https://docs.opentrons.com/v2/new_advanced_running.html) than our normal def run() protocols so you would have to adjust it. 
2. We also have a resource for how to make Jupyter Notebook [work with the rail lights](https://github.com/Opentrons/opentrons/issues/6164) 

##Important Note 
Unless there is a really compelling reason, I wouldn't normally recommend Jupyter notebook to run protocols. You still have to calibrate through the app before running anything and you see "Out of Bounds Errors" that don't really mean it is out of bounds, it is just closer to a boundary condition (example attached below). If you go to **Advanced Settings > Download Logs ** and it shows up in the API.log and Serial.log than it could point to a bigger problem and you can contact us at support@opentrons.com

[image]

mcous, 2021-01-21:
This is expected behavior if you do not, stop the `opentrons-robot-server` first. @alexjoel42's solution should work.

Since this is expected behavior and the system prints a warning to stop `opentrons-robot-server` before trying to interact with the rail lights in Jupyter, I'm going to close this issue. Please feel free to continue posting in this thread with questions!

closed 2021-01-21 by mcous