Opentrons/opentrons #7216: bug: After resuming a protocol, it immediately pauses again, and the cycle repeats

SyntaxColoring, 2021-01-12:
From @ethanfjones (moved from #7167):

> When they hit pause ~30% it stops and then when they resume, automatically resumes to the next step and then pauses again. They need to spam the pause/resume button in order to get it to work ( I know not good)
> 
> [Event 1 .zip](https://github.com/Opentrons/opentrons/files/5739941/Event.1.zip)

SyntaxColoring, 2021-01-12:
One theory was that this has to do with some questionable threading code that appeared in several protocols to flash the deck lights.

I'm a little skeptical of this. I *think* I've seen it happen in [one protocol that doesn't involve threading at all](https://develop.protocols.opentrons.com/protocol/13fd88).

SyntaxColoring, 2021-01-13:
To investigate this bug further, we need help finding clearer steps to reproduce. We tried investigating through the logs, but they turned out to not be enough.

It would be best to have a video of *everything* leading up to the problem—turning on the robot, uploading the protocol, and so on. (And all the relevant protocol and labware files, of course.)

-----

As a start...

@amitlissack has a theory that this can happen if you somehow manage to get two protocols running on the robot at once. Specifically, if one of the protocols is in a delay step. But it’s unclear whether it's possible to get a robot in this state.

Additionally, here are *unconfirmed* steps from some old notes of mine. I happened to stumble upon this problem a while ago while working on something else; I didn't have time to investigate further.

1. The robot is dropping a tip in the trash
2. You pause, then quickly resume
3. Clicking the pause/resume button toggles the label without doing anything
4. Reopen the app and it’s a resume button
5. Click it and the robot continues to the next step, but then freezes. App shows a pause button.
6. Pause/resume now single-steps.

SyntaxColoring, 2021-01-19:
I've confirmed and further narrowed down the reproduction steps.

1. Upload and start running this protocol.
   ```python
   metadata = {
       "apiLevel": "2.8"
   }

   def run(protocol):
       tip_rack = protocol.load_labware("opentrons_96_tiprack_300ul", 8)
       pipette = protocol.load_instrument("p300_multi_gen2", mount="right", tip_racks=[tip_rack])
       
       for column in tip_rack.columns():
           pipette.pick_up_tip(column[0])
           pipette.drop_tip()
   ```
   I don't think the exact protocol matters; this is just what I tested with.
2. Wait for the robot to move to the trash for a drop-tip step.
3. Wait for the pipette mount to lower.
4. To eject the tip, the robot will spend several seconds lowering and then homing its plunger. **While this is happening, click the "Pause" button. Then, quickly click the "Resume" button before the plunger completes its home.**
5. After the plunger home completes, the robot will be physically paused, but the Opentrons App will show a "Pause" button and its clock will be ticking.
6. Continue clicking the "Pause"/"Resume" button to see the problem described in the issue summary.

SyntaxColoring, 2021-01-26:
Note to self: Next step is to repro this with some more logging.

amitlissack, 2021-01-26:
A theory about what is happening.

Clicking pause leads to a call to `opentrons.api.session.Session.pause`. This method does two things:
1) calls `opentrons.hardware.controller.API.pause`. This is the implementation of the method:
```
        async def _chained_calls():
            # sets the execution manager's state to paused. the next call to `wait_for_is_running` will block.
            await self._execution_manager.pause()
            # clears the run_flag `Event` in the smoothie driver.  The next `move` will block on run_flag.
            self._backend.pause()

        # call _chained_calls on the event loop after it's through with whatever motion is currently happening. 
        asyncio.run_coroutine_threadsafe(_chained_calls(), self._loop)
```
2) sets protocol state to paused. This makes its way to the run app to update UI state.

Clicking resume leads to a call to `opentrons.api.session.Session.resume`. This method does two things:
1) calls `opentrons.hardware.controller.API.resume`. This is the implementation of the method:
```
        # Immediately sets the run_flag `Event` in the smoothie driver.  All waiting threads will be awakened.
        self._backend.resume()
        # Set the run state of execution manager to running after the loop is done with whatever it's doing.
        asyncio.run_coroutine_threadsafe(self._execution_manager.resume(),
                                         self._loop)
```
2) sets protocol state to resume. This makes its way to the run app to update UI state.

---

Theory: 
a) In your step 4 `opentrons.hardware.controller.API.pause` is called, enqueueing a call to pause whenever the loop is done with what it is doing (which is a long running task). 
b) You click resume leading to `opentrons.hardware.controller.API.resume`. This is happing BEFORE `_chained_calls` is called because the loop is busy (homing, or moving or whatever).  The run_flag in the smoothie driver is `set` but is then `cleared` when `_chained_calls` is finally called.

PR #7265 (2021-01-28): fix(api): fix race condition causing pause/running state mismatch.
# Overview

Occasionally, during a protocol run, clicking pause then resume will still result in a pause. 

Due to pause being handled in `asyncio.run_coroutine_threadsafe` while `resume` is not, it is possible for the `SmoothieDriver_3_0_0.resume` method to be called before pause actually happens. More detailed [explanation](https://github.com/Opentrons/opentrons/issues/7216#issuecomment-767749590) of the race condition.

closes #7216 

# Changelog

call `self._backend.resume()` twice: 
- immediately to avoid awaken potentiall [... cut]
commit cf62256a (2021-01-28): fix(api): fix race condition causing pause/running state mismatch. (#7265)

closes #7216
closed 2021-01-28 by amitlissack