Opentrons/opentrons #7251: bug: mDNS client in discovery may fail if multiple U2E adapters are present

mcous, 2021-01-23:
## Overview

This bug report requires further investigation, but I encountered a client-side mDNS bug that prevented robot discovery.

## Steps to reproduce

Computer setup

- On macOS 10.15.7 (Catalina)
- Tested with v4.0.0 and v4.1.0 of the robot software
- OT-2 plugged into laptop via USB (thru a USB-A female to USB-C male adapter)
- USB-C hub with Realtek USB-to-Ethernet adapter plugged into another USB-C port, with the Ethernet port unused

While `dns-sd` (built-in macOS mDNS discovery utility) was able to find the OT-2 just find, `@opentrons/discovery-client` reported no mDNS events, and was thus unable to find the OT-2.

The OT-2 was directly connectable using its link-local IP address retrieved via `dns-sd`.

The OT-2 showed up in `@opentrons/discovery-client` as soon as the USB-C hub was removed.

## Further investigation

- Is this reproducible on different setups?
- Is this reproducible on different operating systems?
- What role does https://github.com/mdns-js/node-mdns-js (the JS mDNS library we use) play in this bug?

mattwelch, 2021-04-09:
Consider adding support doc update about internet sharing in MacOS.

mcous, 2021-12-13:
IIRC, this was an issue with the DHCP settings of the adapters

closed 2021-12-13 by mcous