Opentrons/opentrons #7262: bug: When the API has an aspiration volume of 0uL, the OT-2 instead uses maximum pipette aspiration amount.

alexjoel42, 2021-01-26:
# Overview
If you define aspiration volume as 0, then when simulating the protocol will output the maximum volume of the pipette. 

# Steps to reproduce

Try and aspirate 0 uL with a pipette. 

# Example with the GEN2 P300S

```
from opentrons import protocol_api

metadata = {'apiLevel': '2.8'}

def run(protocol: protocol_api.ProtocolContext):
    tr1 = protocol.load_labware('opentrons_96_tiprack_300ul', '1')
    p300_single = protocol.load_instrument('p300_single_gen2', 'left', tip_racks=[tr1])
    reservoir = protocol.load_labware('nest_12_reservoir_15ml', location='2')
    well = protocol.load_labware('opentrons_96_aluminumblock_nest_wellplate_100ul', location='4')
    p300_single.pick_up_tip()
    p300_single.aspirate(0,reservoir.wells()[0])
    p300_single.drop_tip()
```

`

# Current behavior
 
[image]

Aspirates full volume. This happens in simulation with the GEN2 P300S and P300M as well as the GEN2 P1000S. I have run tests on the GEN2 P300S and P300M and they both aspirate full volume.

# Expected behavior

Picking up tip from A1 of Opentrons 96 Tip Rack 300 µL on 1
Aspirating 0.0 uL from A1 of NEST 12 Well Reservoir 15 mL on 2 at 92.86 uL/sec
Dropping tip into A1 of Opentrons Fixed Trash on 12

The software should warn users if the aspiration volume is 0. Not blocking, but it should pop up to make sure that someone doesn't accidentally aspirate full volume. 

# Current work around

Put a very low value 0.0000001 or avoid aspirating 0uL.

mcous, 2021-01-26:
@alexjoel42 FWIW this is documented behavior of `aspirate`: https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.contexts.InstrumentContext.aspirate

> - **volume** (_int_ or _float_) – The volume to aspirate, in microliters (µL). If 0 or unspecified, defaults to the highest volume possible with this pipette and its currently attached tip.

alexjoel42, 2021-01-26:
Thanks for the speedy clarification. This makes sense + is documented. Might be worth talking about if this pops up more, but I have only seen it happen once in 7 or so months so I think we can close this.

closed 2021-01-26 by mcous