Opentrons/opentrons #7263: feat: Any way tom import a custom function in a Python protocol

MarcelRobitaille, 2021-01-26:
# Overview

I have a simple function in a file in the same directory as my protocol using the Python API. I want to keep this function in a separate file since it is shared between multiple files. The obvious `from local_file import my_function` doesn't work because of the way the simulator works. I also thought maybe I could try using importlib, but `__file__` is not defined inside the protocol. I think because the file is read from disk, parsed, and executed instead of imported in the normal Python way.

Is there a recommended way to make local imports inside Python protocols? Maybe you could just define `__file__` when you execute the Python file if path isn't stdin? It should be easy I think, I have done something similar before.

My current workaround is to copy that function between all my files, which is obviously not ideal.

# Implementation details

Maybe [here](https://github.com/Opentrons/opentrons/blob/66ac520402e0e4e9dbf655e508442996f70f81d1/api/src/opentrons/protocols/execution/execute_python.py#L44) you could do something like:

```python
if proto.filename:
    new_globs['__file__'] = proto.filename

exec(proto.contents, new_globs)
```

# Acceptance criteria

- [ ] I should be able to import local functions with either `from local_file import local_function` or using importlib

amitlissack, 2021-01-27:
This is a feature we will include in a future release. It is currently available for beta testing. If you're interested please email beta@opentrons.com.

MarcelRobitaille, 2021-02-03:
A good solution for me for now is to use the beta release. For the simulator, I can use `PYTHONPATH="$(pwd)" opentrons_simulate ...`.

mattwelch, 2021-02-18:
We're working on this capability as part of a future release. We'll close this ticket now. Let us know if you have any other questions.

closed 2021-02-18 by mattwelch