Opentrons/opentrons #7269: bug: Windows users are unable to downgrade from 4.0 to 3.2.1

Gyliann, 2021-01-28:
# Overview

Users have been needing to downgrade to do some 4.0 bugs we’ve seen that have made operating the robot difficult. However, there have been some Windows users, even after going through all downgrading steps, that still have the v4.0 app on their computer. Even if they have 3.2.1 downloaded on the computer, they app still opens to the 4.0 version.

# Steps to reproduce

Download ot2-system.zip for robot software v3.21.2. Don't unzip the file.
In the Opentrons App, go to the Robot tab and make sure you're connected to your OT-2.
Under Advanced Settings, find Update robot software from file and select Browse.
Upload ot2-system.zip.
Wait for your OT-2 to apply the downgrade and restart.
When you reconnect to your OT-2 after it restarts, confirm that it says "v3.21.2" under Server version.
Uninstall Opentrons App v4.0.
Reinstall Opentrons App v3.21.2.

# Current behavior

Both 3.2.1 and 4.0 versions of the app will stay on the computer. The app will open up to the 4.0 version.

# Expected behavior

The 4.0 version of the app should be removed and the customer should be able to use the 3.2.1 version of the app

SyntaxColoring, 2021-01-29:
Someone ran into this internally (maybe):

> on a windows machine if you:
> * Are on 3.21.2 app build, have a desktop shortcut, and have the App pinned to the task bar.
> * Upgrade the App to 4.1.1.
> * The app creates a second desktop shortcut.
> * The shortcuts both open different app versions and the pin to the task bar opens the old app version.

mcous, 2021-02-01:
I've investigated this, and I can definitely reproduce weirdness on my machine. From the symptoms, we've got problems in the Windows registry, at least:

- After the Windows app updates itself, I see 2 entries for Opentrons in "Add or Remove Programs"
- Clicking on one runs the uninstaller but leaves a phantom entry for the other
    - It also may prevent the installer of that version from running

I was able to resolve by taking the following (convoluted) steps:

1. Open the app and get it updated to latest (either via the update mechanism or by running the installer)
2. Verify that you have two "uninstall" entries
3. Run the [3.21.2 installer](https://github.com/Opentrons/opentrons/releases/tag/v3.21.2)
4. Verify that you now have one uninstall entry
5. Use "Add Remove Programs" to run the uninstaller
6. Re-run the 3.21.2 installer
7. Cross fingers?

This looks to be related to some serious buginess in [electron-builder](https://github.com/electron-userland/electron-builder), which has give us registry grief on Windows before

closed 2021-02-26 by asogluizzo