Opentrons/opentrons #7310: Spike: update-server: Updating the boot partition in our update process

sanni-t, 2021-02-09:
# Overview

This issue relates to the RTC support in buildroot initiative: https://github.com/Opentrons/buildroot/issues/108

As a requirement for supporting the RV3028 RTC on our RobotOS, we are updating the OS kernel & Rpi firmware. For a proper deployment of this update through our update-server, changes will need to be made to the update-server process in order to update to boot partition as well.

Spike TODOs:

- [ ] Understand the current filesystem
- [ ] Understand the current update process
- [ ] Figure out what does & does not get updated in the process (So far we know that `config.txt` and the device tree files in the boot partition do not get updated. The wifi interface also doesn't update properly but not sure because of what)
- [ ] Find out what it'll take to update boot partiton. Notes from a conversation with Seth: 
- we may have to try and deliver device trees in the root file system or something. Pack two kernels?
- maybe we can patch up the difference between device trees instead of writing a whole new device tree

sanni-t, 2021-03-09:
Since this issue with our update process has emerged from attempting to update the kernel, we are now looking at ways we can accomplish RTC support without updating the kernel i.e., by backporting the driver instead. That work will be tracked in https://github.com/Opentrons/buildroot/issues/108.

Iceboxing this ticket until a resolution is reached in the above epic.

mattwelch, 2021-03-23:
We are closing this ticket as done, and will write a follow up offboarding ticket to explain why this ticket is moot for the foreseeable future. @sanni-t

closed 2021-03-23 by mattwelch